qpv-research-group/solcore5 #10: Refractive index database

dalonsoa, 2018-05-02:
Solcore includes the SOPRA database of optical properties and has built-in the refractive indices of some materials and alloys. However, the origin of some of this data is not clear (specially the SOPRA database). 

Moreover, the calculation of the optical properties of alloys is hard. Solcore includes a powerful interpolation method based on the selection of a series of critical points in the n and k data, but it is rather tiresome to use, specially if the data to start with the interpolation comes from different sources. 

- Interfacing with the [Refractive index database](https://refractiveindex.info) ([which is stored also in GitHub](https://github.com/polyanskiy/refractiveindex.info-database) will be an excellent way of accessing a huge range of materials correctly referenced. 

- Having an easier to use method for interpolating data and get the optical constants of alloys will also be very useful.

phoebe-p, 2018-09-04:
I have started dealing with the first point above, interfacing with refractiveindex.info, by using the [refractiveindex.info-sqlite code also on GitHub.](https://github.com/HugoGuillen/refractiveindex.info-sqlite)

phoebe-p, 2018-09-13:
Have written functions to interface with the refractiveindex.info database, and to add new materials to Solcore's database so they can be created like 'normal' materials

dalonsoa, 2019-10-04:
Added in 3998e7cc

closed 2019-10-04 by dalonsoa