qpv-research-group/solcore5 #3: Fermi-Dirac statistics

dalonsoa, 2018-05-02:
Solcore's PDD solver, as it is very often the case, uses the Boltzmann approximation. While that is good enough for many applications - and usually faster - there are cases where a more rigorous treatment, using the Fermi-Dirac (FD) statistics, will be useful: highly doped materials, quantum wells, etc. 

Ideally, using FD statistics should be implemented as a modification of the existing solver, rather than a separate solver.

tcaduser, 2018-05-02:
https://github.com/scott-maddox/fdint

tcaduser, 2018-05-02:
I'm confused what f2py actually does.

For my own code, I am using the Joyce-Dixon approximation in double precision.

https://github.com/devsim/devsim/blob/master/src/MathEval/Fermi.cc

dalonsoa, 2018-05-10:
f2py is a Python tool that compiles a Fortran module and makes the functions and variables inside accessible from Python as with any other Python package. You have more information here: https://docs.scipy.org/doc/numpy/f2py/