qpv-research-group/solcore5 #9: Quaternary materials and new materials

dalonsoa, 2018-05-02:
Solcore has built-in most binary and ternary III-V semiconductors with their properties taken from the  [Vurgaftman and Meyer 2001 paper](https://doi.org/10.1063/1.1368156). Quaternary alloys or other material systems (II-VI, CIGS, CZTS, etc.) are simply not considered, which is clearly a limitation.

A system should be included to easily add new materials to the database and calculate the properties of their alloys, wit good, well supported references.

cdeimert, 2022-07-19:
Is there any convenient workaround for quaternaries at this point? My inclination would be to write separate code to calculate all the parameters for each alloy composition, then print the results to a text file and use that to define a new material for each alloy composition in solcore. Are there any pitfalls with such a method?

By the way, if support for quaternaries is ever added, I suggest using the interpolation scheme of Donati et al (https://aip.scitation.org/doi/abs/10.1063/1.1613371), which has an additional bowing parameter for quaternary alloys, independent of the constituent ternary bowing parameters.

lpdsamis, 2024-03-27:
[notice_1.pdf](https://github.com/qpv-research-group/solcore5/files/14770704/notice_1.pdf)