qpv-research-group/solcore5 #11: software description

tcaduser, 2018-05-21:
Hello,

I placed a software description for your software on my website:
https://www.tcad.com/Software.html#open-source-tcad-software

Please let me know if you like any updates to this description:

Solcore https://github.com/dalonsoa/solcore5

A complete semiconductor solver able of modelling the optical and electrical properties of a wide range of solar cells, from quantum well devices to multi-junction solar cells.

dalonsoa, 2018-05-21:
Hello,

Many thanks for this! The description is perfect!

Best,
Diego

----------------------------------
Dr. Diego Alonso Álvarez

EXSS group, Department of Physics
South Kensington campus
Imperial College London SW7 2AZ
London, United Kingdom

web: http://www3.imperial.ac.uk/quantumphotovoltaics
e-mail: d.alonso-alvarez@imperial.ac.uk<mailto:d.alonso-alvarez@imperial.ac.uk>
tel: +44 (0) 20 759 47563

https://www.tcad.com/Software.html#open-source-tcad-software

tcaduser, 2018-05-25:
Done.  I am experimenting with using index terms, so please check out the Index from the main page to see if your entries make sense.
photovoltaic
drift diffusion
TCAD

dalonsoa, 2018-05-25:
Yes, they make perfect sense. However, Solcore entry is linked from [1] except in the case of the index term photovoltaic, which I am not sure if was the expected behaviour. 
Many thanks!

tcaduser, 2018-05-25:
It appears the first item indexed gets the name, and then the other items get a number. Right now i only have your simulator and mine indexed, so the numbers will change.

closed 2018-07-11 by dalonsoa