qpv-research-group/solcore5 #12: Small improvements to the LightSource class

dalonsoa, 2018-06-12:
The LightSource class deals with light sources (the solar spectrum or other spectra) and there are currently two things to improve:

1. Currently, the "total_power" attribute returns the total power by integrating the spectrum in the requested spectral range. It will be convenient to change it so it provides the power of the total spectrum, even if the wavelength range of interest is much smaller. For example, the "total_power" of the AM1.5g solar spectrum should be 1000 W/m2 even if we ask the spectrum just from 300nm to 900 nm because we are modelling a GaAs solar cell. 

2. The laser light source is defined by a gaussian curve... in wavelength. That does not make sense from a physical point of view it it should be defined as a gaussian curve in energy.

YunguiLi, 2020-04-30:
Regarding the TMM solver to calculate the reflextion of a multilayered ARC, it looks like that the absorption is not included, leading to R+T=100%. It looks like that at least in the example. http://docs.solcore.solar/en/master/Examples/example_RAT_of_ARC.html. 
Is it possible to fix this bug? 
Thanks a lot!

phoebe-p, 2020-04-30:
@YunguiLi please open a new issue if you have a problem rather than commenting on a (seemingly unrelated?) different issue!

To answer your question, this is not a bug, the TMM solver does calculate absorption. This is a slightly odd example, where the calculated absorption (which will be in the GaInP) is plotted as "Transmission" because we care about transmission into the GaInP. You can see how to get the absorption per layer in for instance this example: [Coherency example](https://github.com/qpv-research-group/solcore5/blob/devel/examples/coherency_example.py)