qpv-research-group/solcore5 #15: Rigorous models for tunnel junctions

dalonsoa, 2018-06-12:
Solcore has built in a simple resistive model for tunnel junctions, and also a parametric model, defined in terms of peak and valley currents and voltages. It also allows for externally defined tunnel junctions, by getting the tunnel junction IV curve as input. 

More rigorous models will allow to calculate the tunnel current from the layer structure of the junction. Some simple models that could be implemented are:

- Semiclassical Tunnel Diode model - S. M. Sze, Physics of Semiconductor devices (Chapter 8: Tunnel Devices).
- Louarn, K., Fontaine, C., Arnoult, A., Olivié, F., Lacoste, G., Piquemal, F., Bounouh, A., Almuneau, G.: Modelling of interband transitions in GaAs tunnel diode. Semicond. Sci. Technol. 31(6), 06LT01-6 (2016)

Chaz-Photonicity, 2022-09-28:
I've read about this from the documentation but no detail about how to do it.  Pls advise.  Thanks

dalonsoa, 2022-09-28:
Unfortunetly, this is not that well documented - if documented at all :( 

You can have a look at the function that does that here:

https://github.com/qpv-research-group/solcore5/blob/95a2b01dbac7f3feae812709890f0e630cdbbdd6/solcore/analytic_solar_cells/tunnel_junctions.py#L88

Essentially you will need to call this function using as input a junction object (like in the other cases) that has `external_voltage` and `external_current` as attributes, so they can be interpolated. And that's it.

Updating the documentation on this - or adding new models - is a nice, good first issue in case you want to contribute as part of [Hacktoberfest](https://hacktoberfest.com/), by the way.