qpv-research-group/solcore5 #18: KP solver for QWs

dalonsoa, 2018-06-12:
Solcore calculates the properties of QWs by solving the Schrodinger equation in 1D using a simple parabolic approximation. While the properties of the bulk materials and the strain are included, effects like band mixing are completely missed. 

A 4-band KP solver, coupling the heavy and light hole bands, is currently implemented, but it should be done in a more rigorous way. A 6-band or, event better, a full 8-band KP solver for QWs, including the calculation of the absorption of the QWs, will definitely boost Solcore's capabilities.

erw-del, 2020-04-30:
How can i calculate eigen energy values in quantum well , with given barrier height for example 240mev in AlGaAs , normalized and envelop wave function, could you please illustrate

dalonsoa, 2020-04-30:
Please, have a look at [this example](examples/Schrodinger_QW_and_MQW.py). The output of the QM.schrodinger function used towards the end contains all the information related to the band structure, wave functions and energy levels. It is essentially a tuple of dictionaries, and hopefully the keys of the dictionary are self explanatory!

About the structure, you can only define the materials and layer thicknesses involved, not the barrier height. This is calculated automatically for the valence and conduction bands based on the band gaps and electron affinities of the materials involved.

erw-del, 2020-04-30:
Thankyou for your kind reply,
It helped a lot,
I would like to know that is it possible to include electric field to solve
shrodinger equations for hetrostructure,
It will be very kind of you incase you can help me with this notion.

dalonsoa, 2020-05-01:
You can provide `Efield` as input parameter to the schrodinger solver. Please, [have a look at the documentation](http://docs.solcore.solar/en/master/QM/Schrodinger.html#module-solcore.quantum_mechanics.high_level_kp_QW).

erw-del, 2020-05-05:
Thanks for the response,
I want the wave function data in form of arrays"all the points in wave function" for every subband energy calculated . Since, i used "QM.schrodinger" to solve my structure the output provides only few elements as an arrays for every subbands wavelength "the beginning and the end points ",i tried to use"solcore.quantum_mechanics.graphics.prepare_wavefunction_data(x, E, Psi, trim_levels_beyond, linewidth, color, scale, suppress_invert=False, square=False, alpha=1, label='untitled')" but in this situation i do not know how to pass the argument comprising my structure,

I am interested in getting the entire points in form array, for heavy hole,light hole and electrons,
hetrostructure
a simple example of quantum well("single well Hetrostructure") , would help me a lot to understand ,
including the feature of discard_unconfined_energy,
Solcore seems to be very promising platform better understanding of it will help me a lot  
Thanking in anticipation

erw-del, 2020-05-09:
> You can provide `Efield` as input parameter to the schrodinger solver. Please, [have a look at the documentation](http://docs.solcore.solar/en/master/QM/Schrodinger.html#module-solcore.quantum_mechanics.high_level_kp_QW).

what is the unit of electric field,how can i extract entire wave function for every subband energy,"the output provide limited points for every subband"
the results do not match with https://www.nextnano.de/nextnano3/tutorial/1Dtutorial8.htm
and with many other,problems

dalonsoa, 2020-05-09:
Please, stop asking questions in this issue. This was separate topic to improve existing features, not to solve problems. 

Also a couple of things: when asking for help, please provide the exact script you are executing and how you are analysing the data. If you are comparing with other results from a different program and they don’t match, please also provide a figure showing how they differ.

Finally, the arrays provides all the points. It shuts the print statement the one not printing everything. Please, check it’s size or plot it.

erw-del, 2020-05-09:
Thank you very much,
for your help