qpv-research-group/solcore5 #27: Exposing incident angle and coherence list for tmm

P-Tillmann, 2018-11-08:
Hi,

for my research project i need to be able to choose a incident angle for the incoming light different from normal incident. Also, because we want to model thin and thick layers the incoherence options of tmm would be useful. What is your opinion to expose these function to solcore users?

Best,
Peter

dalonsoa, 2018-11-13:
Hi! Sorry for the slow reply.

Yes, exposing the angle of incidence is easy, convenient in the long term and can certainly be done with no effort. 

Now, if the angle is not zero, then it is necessary to consider the two polarisations of the light (s and p) and then put the results together - assuming you are interested in unpolarised light. This has to be done for the reflected and absorbed light vs depth. This can certainly be done (the TMM solver can calculate it) but it is s little bit more work - and twice the computation time, in case that matters. 

In any case, by all means, go ahead with it! 

Let me know if you need any assistance or more information on how things work internally. @phoebe-p  is the one with most experience with the optical models, so she might have a opinion on this, too. 

Best wishes

phoebe-p, 2018-11-13:
I have recently made some changes to the TMM solver which means you can now select the input angle. It still assumes unpolarised light, so it calculates both s and p polarization and then averages them. This version is currently in the 'tmm_correction' branch, but will hopefully be merged into the main branch in the next few weeks.

P-Tillmann, 2018-11-13:
Thanks for your replies. If the incident angle has already been implemented even better for me :-)
What do think about the coherence list feature of the tmm solver? If i understand the documentation of tmm right it is not possible to calculate the depth dependent absorption in an incoherent layer. Is this needed for some of the solar cell solvers? If so, the depth dependence could maybe be approximated with Lambert Beer, which should be reasonable for thick incoherent layers.

phoebe-p, 2018-11-13:
Yes, we need the depth-dependent absorption for some of the electrical solvers. In thin layers, it can deviate a lot from normal Beer-Lambert absorption but you are right that in thick layers they are very similar. There is also a Beer-Lambert optical module in Solcore for quick calculations in such situations. There’s no reason we can’t implement the coherency list feature, and use Beer-Lambert absorption in the incoherent layers (in fact I am working on something similar where if there is a thick substrate/final cell in a multi-junction stack, it is treated incoherently using BL law. This avoids memory errors and issues with unphysical oscillations. This is also in the tmm_correction branch).

P-Tillmann, 2018-11-13:
Ok, thanks for all the information. 

Does it make a difference if:
tmm_coherent -> substitute incoherent layers with LB (what you are working on if i understood correctly?)
tmm_partly_incoherent -> substitute incoherent layers with LB

Otherwise there will be no need to expose tmms own coherence list.

phoebe-p, 2018-11-20:
As far as I can see, for many situations there won't be a difference (you will just have to assume Beer-Lambert absorption in the incoherent layer, regardless of the approach). However, there is a case which can be relevant, where I think the tmm_partly_incoherent approach makes sense: it is when you have further thin layers beneath a thick layer you want to treat incoherently (so a coherent/incoherent/coherent layer stack). In my approach currently, once you hit a very thick layer, that layer and any layers below it are treated with a Beer-Lambert approach (so can only do coherent/incoherent/incoherent... etc.). I am not sure how the partly incoherent treatment in the TMM module deals with those cases...

P-Tillmann, 2018-11-23:
The tmm module is grouping consecutive coherent layers and treat them as if they were the first layers (with reduced intensityies of course). I'm currently finalizing my approach and will then commit it. Then you can decide what makes sense to pull into master and were you want to do it differently.

phoebe-p, 2019-07-15:
This issue was address by #43 .

closed 2019-07-15 by phoebe-p