qpv-research-group/solcore5 #33: 6000 max meshpoints PDD solver

jbuencuerpo, 2019-03-01:
In the wiki it is said that the absolute limit is 6000. If the structure is too thick, the code fails silently.
Why 6000 is the limit?

dalonsoa, 2019-03-04:
In the Fortran code, the maximum size of the arrays is hardcoded - and set to 6000 -. Originally, when it was done 5 years ago, it was just a question of convenience and of simplifying the code, but it can be done better and passed that value dynamically. 

As an intermediate solution, I would suggest you play with the [mesh size and growth rate](http://docs.solcore.solar/en/master/Solvers/DriftDiffusionUtilities.html#setting-different-aspects-of-the-solver) in the solver options to try to keep the total number of mesh points below 6000. 

Keeping aside that this can be changed, 6000 mesh points is really a huge number for a 1D calculation. I've simulated solar cells with [up to 100 strain balanced QWs](url
https://doi.org/10.1109/PVSC.2014.6925530) and didn't reach that value. I'm curious, what are you trying to model?

jbuencuerpo, 2019-03-12:
I was playing with the tutorial for the tandem with QWs, no real structures, and the mesh size. The code fails silently  using for example a jupyter notebook, and there is no Warning in the python code. I think it will be helpful to raise an Error in python.

I am not literate in Fortran, I have been able to increase the hardcode number to 8000 in the fortran code, but a dynamic upper number will be desirable.

dalonsoa, 2019-03-13:
I completely agree. I've changed the title of the issue and labelled it as enhancement.

jmllorens, 2019-03-13:
I have started to implement the dynamic version of pdd.  This is the branch in my fork:
[https://github.com/jmllorens/solcore5/tree/dynamic_dd](dynamic_dd)

jbuencuerpo, 2019-03-14:
I have created this small example for a GaAs SJ. Increasing the meshpoints crash it silently.

[https://gist.github.com/jbuencuerpo/436d0791a931a0e727b1921c841a2c98](https://gist.github.com/jbuencuerpo/436d0791a931a0e727b1921c841a2c98)