qpv-research-group/solcore5 #34: Quasi3D solver improvement

ilomdez, 2019-03-12:
As I am not sure of how important this may be to the Solcore community I'll write a general idea of what could be improved. If something is of particular interest a more accurate issue can be opened.

General improvements:
  - Allow to define the height and the width of each row and column to improve accuracy/speed
  - Allow to save the outputs(.out and .raw). In my experience it is important to keep this files or a t least set an option to make solcore save them. The .out might have some warnings, so as long as they are not errors the simulation will run without the user noticing the warning. Allowing the .raw to be saved opens the possibility for the user to use their own data reader/plotter.
  - Allow to sweep a variable. Of course this can be done in python, sweeping a variable and then executing the quasi3D. However it is usually faster to let Spice do this, as it has the circuit already built and it does not have to be completely rebuild for each iteration
  - Avoid printing the nodes voltages as default. Printing the values take a lot of time for large circuit and it is unnecesary when only the I-V curve is needed.
  - Allow to get the current going through a device in particular (e.g. current going through J01 diodes)
  - Allow to get the voltage difference of a junction. This will allow to obtain the EL or PL of a junction straightforward.

Stand-alone Improvements:
  The integration of the Spice solver with the rest of the modules is awesome and really useful. However, it set some constraints that limit the quasi3D potential and developing a quasi3D that does not depends on the rest of the solcore structure might be useful as well.
Of particular importance would be the to allow user-defined circuits to:
         a) Take into account non-homogeneous effects (such as the perimeter recombination)
         b) Allow the use of other devices (capacitors, reverse breakdown diodes) that are usually not needed to model the standard behaveiour of a solar cell, but that may be useful for certain scenarios (EQE measurement simulation for instance)
         c) Simplify circuits - for instance you don't need lateral resistances at the edges of each junction in a multijunction solar cell under certain circumstances. This allow to speed up the spice solver and to solve much bigger circuits.   
 
Again, if this is not of particular importance for solcore, please let me know and I'll try to find an open issue that suits me.

Thank you

dalonsoa, 2019-03-13:
I think it is a really useful feature for the PV community, specially for those working at higher, full solar cell level - maybe not so much for those in semiconductor physics. So, in summary, anything that makes the Solcore/SPICE interface faster and more versatile, is more than welcome. 

I specially like - and I'm sure @iclned would agree - the possibility of including luminescence coupling. I know it could be done with dependent current sources (I think that it'll be current dependent current sources, actually) but never got time to implement it. That will be an excellent improvement!

Keeping aside all the other enhancements - that I think are all excellent - whatever is changed should keep backwards compatibility, so the solver can be used in its simplified way, as it is now, if needed.

ilomdez, 2019-03-13:
Great, that's good to know!

Regarding the current sources, I think you can actually use both depending on how elegant you want to be. I mean, subcell luminescence depends on exp(V*q/kT -1), so you can use the voltage and calculate the exponential or use the current through the J01 diode (which will follow the same exponential presumably). Probably the current option is faster while the voltage is slightly better if you are going to mess with the circuit devices. Anyway, changing the type of current source should be easy enough, so we do not need to worry about that yet.

I'll keep an eye on the backwards compatibility while thinking how the enhancements can be added to solcore.