qpv-research-group/solcore5 #39: Control Verbosity

jbuencuerpo, 2019-03-28:
I didn't find it in the docs, it is possible to minimize the Verbosity? Solving one cell is 5-10 lines of outpud and also the PDD solver output appears in the std but not in the notebook for example.

dalonsoa, 2019-03-29:
Controlling the verbosity hasn't been a priority, so I am afraid there are little options there, at the moment. 

Ideally, we will like to implement a [proper logging system](https://docs.python.org/3/library/logging.html) and not just printing stuff. This way, you can control the degree of verbosity by selecting the logging level, from DEBUG (highest) to CRITICAL (lowest), and send things to a log file if needed. 

Now, making this work with the PDD, which send things directly to the standard output might be trickier. It might be possible to use [something like this](https://stackoverflow.com/questions/977840/redirecting-fortran-called-via-f2py-output-in-python) combined with a python callback function from fortran.

jbuencuerpo, 2019-03-29:
Ok, yes I was checking the logging lib. Maybe with the PDD it will be better on the compiling  step add a debug option, so by default it doesn't print output, only on fails.