qpv-research-group/solcore5 #44: Unable to install solcore with pdd

vmercel, 2019-08-16:
when i try installing solcore using:

$ python setup.py install --with_pdd

**I get the following warnings:**

reading manifest template 'MANIFEST.in'
warning: no files found matching 'solcore/poisson_drift_diffusion/ddModel.cpython-36m-darwin.so'
warning: no files found matching 'solcore/poisson_drift_diffusion/ddModel.cpython-36m-darwin.so.dSYM/Contents/Info.plist'
warning: no files found matching 'solcore/poisson_drift_diffusion/ddModel.cpython-36m-darwin.so.dSYM/Contents/Resources/DWARF/ddModel.cpython-36m-darwin.so'
warning: no files found matching 'solcore/material_data/refractive_index_DB/__init__.py'
warning: no files found matching 'solcore/material_data/refractive_index_DB/dbmaterial.py'
warning: no files found matching 'solcore/material_data/refractive_index_DB/dboperations.py'

the PDD fails always even with
$ pip install solcore --install-option="--with_pdd"

My OS is ubuntu in a 64bit HP notebook
the complete installation log is hereby attached
Any help will be highly appreciated.

phoebe-p, 2019-08-20:
Hi! I've just tried the following steps in Ubuntu 16, Python 3.5 and also in Ubuntu 18, Python 3.6:

git clone https://github.com/qpv-research-group/solcore5.git
cd solcore5
sudo python3 setup.py install --with_pdd

I get exactly the same warnings about "no files found", however, the installation does complete (final line of the output is "Finished processing dependencies for solcore==5.6.0.dev0").

Now, after the installation, the first time I open Python and try "import solcore.poisson_drift_diffusion", I still get a warning that the PDD solver is not available - however, if I close Python and open a new process, and try the same thing again, it works with no warnings or errors. So every step on the command line to test if it is working:
```

cd examples
python3
import solcore.poisson_drift_diffusion

```
[get output with warning about PDD not being available]
```

exit()

python3 MJ_solar_cell_tutorial.py

```
This example should give some output which shows the PDD is working (this should be obvious - the first thing it prints is information about the mesh).

Let me know if you have any other issues, and if so which warnings you are receiving. Do you get any errors during the installation or only warnings?

vmercel, 2019-09-01:
Thanks, It is all good now.

closed 2019-09-01 by vmercel