qpv-research-group/solcore5 #45: Bottle-neck in OptiStack.get_indices()

Beadsworth, 2019-09-28:
Hello,

I have had some trouble using the OptiStack objects to get TMM results.  I am modeling VCSELs and DBRs -- which often require hundreds of epi layers, often with the same materials repeated many times.

It seems OptiStack.get_indices() creates a bottle-neck that slows down calculations.  Please note that I am initializing the OptiStack using the Materials and Layers Objects.

It seems that, for every call to OptiStack.get_indices(), there is a file-open operation in Material.n_interpolated().  Is this true?

For Material objects, I have tried caching the results of .n_interpolated() to speed up OptiStack .get_indices().    From this, I see the calculation duration is dropped significantly, and that this effect is magnified for larger repetitions.

Below, I have attached my results.  I modeled a high-index/low-index AlGaAs DBR with a varying amount of mirror-pair repetitions.  "Speed-up" is the uncached_calc_time / cached_calc_time.  I find caching the results of Material.n_interpolated() for each material often speeds up the calculation of OptiStack.get_indices() by at least a factor of 10X.

Is this behavior expected?  Right now, I am under the impression that materials objects, throughout their lifetime, are opening material database files multiple times.

[image]

[model_test.txt](https://github.com/qpv-research-group/solcore5/files/3665842/model_test.txt)

dalonsoa, 2019-10-04:
Thanks for pointing this out! 

You're absolutely right. As Solcore was originally designed for solar cells, with just a few tens of layers, at most, and normally of different materials, we hadn't notice this, but in your case this is certainly an issue. 

In principle, loading data from file is an operation already cached. You can see the [`lru_cache` decorator](https://docs.python.org/3/library/functools.html#functools.lru_cache) in `material_system.py` doing precisely this: 

```python
            @lru_cache(maxsize=1)
            def load_n_data(self):
                if len(self.composition) == 0:
                    wl, n = self.data.load_n()
                    self.n_data = np.vstack((wl * 1e-9, n))

            @lru_cache(maxsize=1)
            def load_k_data(self):
                if len(self.composition) == 0:
                    wl, k = self.data.load_k()
                    self.k_data = np.vstack((wl * 1e-9, k))
```

So probably the bottleneck is in the interpolation itself. The solution will be to use the same `lru_cache` decorator in the `n_interpolated` and `k_interpolated` functions (what you have done with a custom function, but using Python's built-in functionality).

I would suggest you check if using this works for you and open a pull request with the modifications.

PS: If you are participating in the [Hacktoberfest](https://hacktoberfest.digitalocean.com), you can win a t-Shirt :) .

Beadsworth, 2019-10-04:
Will @lru_cache work if the wrapped function does not a return value?

dalonsoa, 2019-10-08:
Yes, it does. If the inputs match a cached configuration, the function simply is not run at al. You can check it by executing the following example:

```python
from functools import lru_cache

@lru_cache(1)
def my_fun(a):
    print(a)

my_fun(3)
my_fun(3)
my_fun(4)
```

Number 3 is printed only once since the second time the function was cached and is not executed.

jaibhageria, 2019-10-25:
@Beadsworth @dalonsoa Has this issue been solved?

dalonsoa, 2019-10-25:
In principle, no, it hasn't. 

As I mentioned above, I think that the solution is as as to use the same `lru_cache` decorator in the `n_interpolated` and `k_interpolated` functions. It will be good to do some checks timing the gain of having this, as @Beadsworth did with his custom-made cache.

phoebe-p, 2020-02-22:
PR #79 addresses this issue.

closed 2020-02-25 by phoebe-p