qpv-research-group/solcore5 #46: Change examples to Jupyter Notebooks

dalonsoa, 2019-10-04:
Solcore has [a few examples](https://github.com/qpv-research-group/solcore5/tree/devel/examples) about how to use several of its features, but these are just plain text python scripts. It will be useful to port them to Jupyter Notebooks, with several benefits:

- Separating code and comments/explanations in separate markdown cells will improve clarity of what is going on where.
- The examples can be run in Binder, without the need of installing anything (with some caveats). 
- Jupyter is becoming the interface of choice in many research environments, specially teaching, so it will be more familiar. 

If this is done, it will be useful to create a pre-commit hook so the output of the notebooks is cleared before any commit. See for example [nbstripout](https://github.com/kynan/nbstripout).

federikovi, 2019-10-04:
Hi, 
I created a [Jupiter notebook](https://github.com/federikovi/solcore5/blob/devel/examples/Light_sources_example.ipynb)
 for the Light_sources example, honestly I don't understand how to make the pre-commit. 

thank you!

dalonsoa, 2019-10-04:
Fantastic! Well done! Not to worry! One step at a time. 

Now you should open a pull request from your fork to this one, so you contribution can be, eventually, added to the main repository. 

If you have never done it, just follow [these instructions](https://help.github.com/en/articles/creating-a-pull-request-from-a-fork) and let me know if you have any doubt.

bigpythonimish, 2019-10-05:
Hi, I have added a jupyter-nb for substrate_examples.

emmanuel-carreira, 2019-10-10:
Hi, i added a juptyter notebook for the Absorption_profile_AlGaAs_GaAs_structure.py example in pr #54

closed 2020-10-01 by dalonsoa