qpv-research-group/solcore5 #47: Improve tests - NOT DOMAIN SPECIFIC

dalonsoa, 2019-10-04:
Solcore has a very limited test suite and those test that exists are functional tests rather than unit tests. Pretty much all Solcore modules need further testing. 

Some modules (and improvements) that could have tests and that do not require specialised solars cells or semiconductors knowledge - just some knowledge of maths - are:

Improvements:
- [ ] Add a `conftest.py` file to group a few things used consistently across several tests. 
- [ ] Move to an external file the expected result of those tests where this output is a big array (like [this one](https://github.com/qpv-research-group/solcore5/blob/df87bf8cff64043bda31f73f14239673cb5b3932/solcore/tests/test_absorption.py#L68)).

Some modules that need tests (not too much domain specific knowledge required):
- [ ] Absorption calculator package: [dielectric_constant_models.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/absorption_calculator/dielectric_constant_models.py), [tmm_core_vec.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/absorption_calculator/tmm_core_vec.py)
- [ ] Light source package: All modules, but specially [light_source.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/light_source/light_source.py). 
- [ ] [structure.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/structure.py)
- [ ] [singleton.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/singleton.py), [source_managed_class.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/source_managed_class.py) and [state.py](https://github.com/qpv-research-group/solcore5/blob/devel/solcore/state.py) (can re-use some examples at the bottom of the files)

Many (most!) other modules also require tests, but they are, most likely, domain specific.

PieceMaker, 2019-10-04:
I've been wanting to brush up on Python so I'll poke this some over the weekend.

phoebe-p, 2019-10-10:
I will work on tests for tmm_core_vec

PieceMaker, 2019-10-13:
I'll see if I can test `structure.py`.

phoebe-p, 2019-10-17:
It's not on the list, but I will work on tests for analytic_solar_cells.depletion_approximation, since I want to get a better understanding of how it works anyway.

closed 2020-10-01 by dalonsoa