qpv-research-group/solcore5 #52: Move tests outside Solcore package

dalonsoa, 2019-10-10:
The canonical place for the tests is outside the Solcore package itself. It will be useful to move it so in the root directory we have:

- docs
- examples
- solcore
- tests
- (setup.py and other files)

This change must be done with care to avoid causing trouble to those working on writing tests.

yuriy-os, 2019-10-15:
Hi! Can I help you with that?

dalonsoa, 2019-10-15:
Sure! There might be some conflict as @phoebe-p and @PieceMaker are both writing tests, but I guess it can be done step by step in those test files they are not working on, yet.  

I would suggest the following approach:

1. You move those tests they are not working on and open a PR.
2. I check and, eventually, merge the PR.
3. @phoebe-p and @PieceMaker get those changes and move the tests they are working on to the new location. 

How does it sound?

yuriy-os, 2019-10-15:
That sounds great! I've found this issue https://github.com/qpv-research-group/solcore5/issues/47. As far as I can tell, I need to move the following tests:
- dielectric_constant_models.py
- singleton.py
- source_managed_class.py
- state.py

Tests `tmm_core_vec` and `structure.py` are in development. That's coorrect?

dalonsoa, 2019-10-15:
Not really. Those files you mention are modules that need tests, but the corresponding test files don't exists yet. 

The end goal is to move `solcore/test/*` to `test/*`. As far as I can tell, there is no file there that, at the moment, is being edited in a PR (well, there is one in #57 but it is mine and can easily edit it).

So go ahead and move everything as soon as possible before some adds another test file!

phoebe-p, 2019-10-15:
Just to confirm, I am writing the tests for tmm_core_vec right now, but they are in a new file. So I can easily pull and move it to the correct folder if this change happens before I finish writing the tests.

dalonsoa, 2019-10-15:
Done! Please, get the latest version of the `devel` branch before adding more tests to Solcore, so they are added to the correct place.

closed 2019-10-17 by dalonsoa