qpv-research-group/solcore5 #55: Type Annotations

dalonsoa, 2019-10-10:
[Type annotations](https://docs.python.org/3/library/typing.html) are an invaluable resource to inform the user what a function or method expect as inputs and will provide as output. It can also help to identify problems using tools like [mypy](http://www.mypy-lang.org). 

Solcore lacks completely of type annotations. 

Annotate everything is an enormous undertaking and it might be very complicated at times, with confusing or custom made objects serving as inputs/outputs. But it can be done bit by bit. Any progress along these lines will be very helpful for the long term sustainability of Solcore.

mhrownaghi, 2019-10-10:
I want to fix this issue.

dalonsoa, 2019-10-10:
You're most welcome!! 

It will be helpful to reduced the disruption to the work of other people also contributing if you tackle the work opening one PR per module you modify.

Abelarm, 2019-10-10:
Hi I would also work on this, if there is "space" for me.
@mhrownaghi maybe we can split up the work. 

What do you think @dalonsoa ?

dalonsoa, 2019-10-10:
Solcore has plenty of modules for both of you - and even more people! Just decide among yourself who is going to work on what. 

I would say that the `analytic_solar_cells`, `poisson_drift_difussion` and `quantum_mechanics` packages are the hardest, by far, so probably you should start by another one. The modules in the root directory are probably the best to start with as they are the core packages and, hopefully, the simplest (keeping aside the `solar_cell_solver` which uses the three above.

mhrownaghi, 2019-10-10:
I will work on modules:
config_tools.py
crystals.py
interpolate.py
science_tracker.py
singleton.py
smooth.py

Artyko, 2019-10-12:
Hi, I'm fairly new in contribution, do you mind if I take up `graphing` module?