qpv-research-group/solcore5 #84: Specifying arguments for 2D junction solver

iclned, 2020-03-27:
The 2-diode (2D) junction solver is presently configured using Eg to set J01 and a radiative efficiency to set J02.  This is quite limiting since it is often useful to specify J01 and J02 explicitly, *not* calculate them from Eg and radiative efficiency!  
It should be possible to set J01 and J02 directly in the Junction () definition.

dalonsoa, 2020-03-27:
This is not really correct. As shown from [this line](https://github.com/qpv-research-group/solcore5/blob/b1b414f734f79f08ea338ffcfef563c4972f27c4/solcore/analytic_solar_cells/diode_equation.py#L47), if there's no need to specify `Eg`: you can define the junction using `j01` and `j02`. The bandgap is needed only if the reference temperature is different than the working temprature. 

If you create a junction providing `j01` and `j02`, do you get and error? If so, could you post it?

iclned, 2020-03-31:
`import numpy as np

from solcore.solar_cell import SolarCell
from solcore.light_source import LightSource
from solcore.solar_cell_solver import solar_cell_solver
from solcore.structure import Junction
from solcore.spice.pv_module_solver import spice_junction
from solcore.spice.pv_module_solver import solve_pv_module

# Code to use the internal SolCore solver
def dsolver( concX,iscarray,j01array ):
    T=273+25
# First we define the properties of the MJ solar cell that the solar module is made of. We use junctions of kind 2-diode
    ge_junction = Junction(kind='2D', T=T, reff=1, jref=300, Eg=0.66, A=1, R_series=0.000, R_shunt=1e14, n=3.5,jsc=10*iscarray[2]*concX, j01=j01array[2])
    ingaas_junction = Junction(kind='2D', T=T, reff=1, jref=300, Eg=1.4, A=1, R_series=0.00, R_shunt=1e14, n=3.5,jsc=10*iscarray[1]*concX, j01=j01array[1])
    ingap_junction = Junction(kind='2D', T=T, reff=1, jref=300,Eg=1.9, A=1, R_series=0, R_shunt=1e14, n=3.5, jsc=10 * iscarray[0] * concX, j01=j01array[0])
#Assemble a solar cell from these juctions
#    my_solar_cell = SolarCell([ingap_junction, ingaas_junction, ge_junction], T=T, R_series=0.0, area=0.1)
    my_solar_cell = SolarCell([ingap_junction],T=T, R_series=0.0, area=1e-4)
    wl = np.linspace(350, 2000, 301) * 1e-9
    light_source = LightSource(source_type='standard', version='AM1.5d', x=wl, output_units='photon_flux_per_m',
                               concentration=concX)

    # options = {'light_iv': True, 'wavelength': wl, 'light_source': light_source}
    V = np.linspace(0, 5, 500)
    solar_cell_solver(my_solar_cell, 'iv',
                          user_options={'T_ambient': T, 'db_mode': 'top_hat', 'voltages': V, 'light_iv': True,
                                        'internal_voltages': np.linspace(-6, 5, 1100), 'wavelength': wl,
                                        'mpp': True, 'light_source': light_source})

    efficiency=my_solar_cell.iv["Eta"]
    pmax=my_solar_cell.iv["Pmpp"]
    ff=my_solar_cell.iv["FF"]
    voc=my_solar_cell.iv["Voc"]
    isc=my_solar_cell.iv["Isc"]

    return [concX,isc,voc,ff,efficiency,pmax]

print(dsolver(200,[14,14,17],[5E-27,6E-19,4.93E-6]))
print(dsolver(200,[14,14,17],[5E-26,6E-18,4.93E-5]))

`

iclned, 2020-03-31:
Here's the code.  Changing j01 makes no difference to the results:

```
import numpy as np

from solcore.solar_cell import SolarCell
from solcore.light_source import LightSource
from solcore.solar_cell_solver import solar_cell_solver
from solcore.structure import Junction
from solcore.spice.pv_module_solver import spice_junction
from solcore.spice.pv_module_solver import solve_pv_module

# Code to use the internal SolCore solver
def dsolver( concX,iscarray,j01array ):
    T=273+25
# First we define the properties of the MJ solar cell that the solar module is made of. We use junctions of kind 2-diode
    ge_junction = Junction(kind='2D', T=T, reff=1, jref=300, Eg=0.66, A=1, R_series=0.000, R_shunt=1e14, n=3.5,jsc=10*iscarray[2]*concX, j01=j01array[2])
    ingaas_junction = Junction(kind='2D', T=T, reff=1, jref=300, Eg=1.4, A=1, R_series=0.00, R_shunt=1e14, n=3.5,jsc=10*iscarray[1]*concX, j01=j01array[1])
    ingap_junction = Junction(kind='2D', T=T, reff=1, jref=300,Eg=1.9, A=1, R_series=0, R_shunt=1e14, n=3.5, jsc=10 * iscarray[0] * concX, j01=j01array[0])
#Assemble a solar cell from these juctions
#    my_solar_cell = SolarCell([ingap_junction, ingaas_junction, ge_junction], T=T, R_series=0.0, area=0.1)
    my_solar_cell = SolarCell([ingap_junction],T=T, R_series=0.0, area=1e-4)
    wl = np.linspace(350, 2000, 301) * 1e-9
    light_source = LightSource(source_type='standard', version='AM1.5d', x=wl, output_units='photon_flux_per_m',
                               concentration=concX)

    # options = {'light_iv': True, 'wavelength': wl, 'light_source': light_source}
    V = np.linspace(0, 5, 500)
    solar_cell_solver(my_solar_cell, 'iv',
                          user_options={'T_ambient': T, 'db_mode': 'top_hat', 'voltages': V, 'light_iv': True,
                                        'internal_voltages': np.linspace(-6, 5, 1100), 'wavelength': wl,
                                        'mpp': True, 'light_source': light_source})

    efficiency=my_solar_cell.iv["Eta"]
    pmax=my_solar_cell.iv["Pmpp"]
    ff=my_solar_cell.iv["FF"]
    voc=my_solar_cell.iv["Voc"]
    isc=my_solar_cell.iv["Isc"]

    return [concX,isc,voc,ff,efficiency,pmax]

print(dsolver(200,[14,14,17],[5E-27,6E-19,4.93E-6]))
print(dsolver(200,[14,14,17],[5E-26,6E-18,4.93E-5]))

```

dalonsoa, 2020-04-01:
The problem is that you also provide the radiative efficiency. When that happens, the program assumes you want to go fully fundamental and calculates `j01` using the detail balance. That’s why it raises an error. 

Have a look at [this file](solcore/analytic_solar_cells/diode_equation.py). The problem is not complicated to solve and should require either adding another branch to the off statement or - even better - reformulating how the two diode kind of junction is defined, including explicit inputs, as we discussed the other day.

dalonsoa, 2020-04-01:
In summary, I believe we have the following set of possible inputs for the Junction (I think):

- `j01`, `j02`, `n1`, `n2`, `Rseries`, `Rshunt`
- `j01`, `reff`, `n1`, `n2`, `Rseries`, `Rshunt`
- `Eg`, `reff`, `n1`, `n2`, `Rseries`, `Rshunt`, `T`

At the moment, we’re missing the second option.