qpv-research-group/solcore5 #91: Problem with Solcore interpreting custom materials as alloys

phoebe-p, 2020-03-31:
The function SolcoreMaterialToStr in structure.py checks whether a material is an alloy in this way:
```

`    material_string = material_input.__str__().strip('<>').split(" ")
    material_name = material_string[0].strip("'")
    composition = {'material': material_name}

    if len(material_name) > 4:

        material_composition = material_string[2].split("=")
`

```
i.e. by checking whether the name of the material is more than 4 characters long (if len(material_name) > 4). This works for differentiating e.g. GaAs vs. InGaP but creates a problem when a custom material is added. E.g. if you add a new custom material called 'Ag_new' to the database, this function will think it is an alloy and then throw an error because no 'composition' is specified:

```
  File "/home/phoebe/Documents/solcore5/solcore/structure.py", line 145, in SolcoreMaterialToStr
    print(material_string[2].split("="))
IndexError: list index out of range
```

At present, this issue can be "resolved" by not defining any new materials with a name more than 4 characters long, but this in general does not seem like a robust way to check whether something is an alloy, it is only a coincidence of the name of the III-V alloys that this works at all. So it seems necessary to find some other, or possibly additional way of specifying whether a material is an alloy or not.

dalonsoa, 2020-03-31:
The easiest solution is to check if the material in the database has “x” as option. Only alloys have an “x”.

dalonsoa, 2020-03-31:
In other words, checking:

if “x” in ParameterSystem().database.options(material_string) then alloy true.

And then we can look for the “parents” of the alloy.

phoebe-p, 2020-03-31:
Excellent, will assign this to myself

phoebe-p, 2020-04-08:
Fixed by #97

closed 2020-04-08 by phoebe-p