qpv-research-group/solcore5 #92: Setting distance spacing for whole stack or individual layers in absorption profile

phoebe-p, 2020-04-01:
Currently, if not specified, the spacing for absorption profile calculations is set automatically as:

``` 
    if options.position is None:
        options.position = np.arange(0, solar_cell.width, 1e-10)

```

in prepare_solar_cell. This spacing of 0.1 nm is good for e.g. most III-V cells but isn't reasonable for indirect gap materials like Si, where layers are often 30 um or more thick. This is a problem for e.g. a triple-junction cell with Ge as the bottom cell, or cells with EVA coatings, cover glass etc. which needs to be included in the model for optical reasons. In this case, such small spacing is both unnecessary for resolving the depth-dependent behaviour in the thick layers and also leads to extremely large matrices which can cause overflow errors etc. and lead to very long calculation times. E.g. for a 4J cell with 100 um Ge as the bottom cell, an example had computation times scaling as follows:

- 1e-8 (10 nm) spacing: 50 s
- 1e-9 (1 nm) spacing: 70s
- 1e-10 (0.1 nm) spacing: 328 s (this is currently the default spacing!)

It is possible to specify the 'position' vector explicitly through the options, but this requires the user to generate a list of z-points before calling the solver. Instead, it would be useful to also be able to specify the spacing by either:

- [ ] Supplying a single number (e.g. '1e-9') as the 'position' option, to use that as the spacing throughout
- [ ] Supplying a list of numbers equal to the number of layers in the cell, and the spacing for each layer is taken from the corresponding list entry automatically. This way, smaller spacing can be used for highly-absorbing III-V layers and larger spacing can be used for more weakly absorbing materials.

dalonsoa, 2020-04-01:
I think this is an excellent idea: it makes life easier for people who wants more control over the process (and speed up calculation!). It has the beauty that requires little extra coding. I'll tag this as "good first issue".

phoebe-p, 2020-04-03:
Fixed by #96.

closed 2020-04-03 by phoebe-p