qpv-research-group/solcore5 #109: PDD solver: error in the documentation on how to access the output

dalonsoa, 2020-05-17:
The documentation for the PDD solver states that the output is provided as a nested dictionary. However, that is not correct: it is provided as a `State` object with the different keys accesible as attributes. For example:

```python
solar_cell[junction_index]['QE']['EQE']
```
should be:
```python
solar_cell[junction_index].qe.EQE
```
[The examples are incorrect](http://docs.solcore.solar/en/master/Solvers/DriftDiffusionUtilities.html#drift-diffusion-utilities) and so is the [description of the dictionary](http://docs.solcore.solar/en/master/Solvers/DriftDiffusionUtilities.html#output-dictionary) at the end.

Abelarm, 2021-10-19:
Great I run this command:
python -m numpy.f2py DDmodel-current.f95-m ddModel -h ddModel.pyf

and it created a .so and a .pyf and now it works!!

dalonsoa, 2021-11-26:
I'm sorry it took me so long to answer. Yes, this is really disconcerting and, honestly, very confusing for the users being the output so inconsistent. 

I think that the only solution - and is not really a solution - that does not break backwards compatibility is to go through all the relevant functions in the [PDD solver](https://github.com/qpv-research-group/solcore5/blob/develop/solcore/poisson_drift_diffusion/DriftDiffusionUtilities.py) and document exactly what their outputs are, event if these are weird and inconsistent. 

Eventually, we need to put some order here and refactor all the outputs, so things make sense, but I don't think we are there, yet, I am afraid.