qpv-research-group/solcore5 #121: Can no longer set LightSource concentration after most recent changes

phoebe-p, 2020-05-26:
Setting the concentration with the 'conc' argument in a LightSource object, and then passing that object through the light_source option to solar_cell_solver, has no effect on the calculated efficiencies etc.; the concentration is always set to 1 internally. Presumably relates to changes in the LightSource __init__ function due to recent bug fixes.

Relatedly, I guess we should add a test for different spectrum concentrations!

dalonsoa, 2020-05-26:
Yep. Sorry I missed that one. It can be set [when calling the spectrum](https://github.com/qpv-research-group/solcore5/blob/234ab5e5cae16de21da07bdeca7dd214a68b41cb/solcore/light_source/light_source.py#L97). In any case, the keyword should be `concentration` not `conc`.

I'll update the constructor as soon as possible.

phoebe-p, 2020-05-26:
Yes, sorry, just realized I typed it wrong in the issue but I was using 'concentration' rather than 'conc' in the actual code I was trying to run (which worked last week).

closed 2020-05-26 by dalonsoa