qpv-research-group/solcore5 #124: Cannot add custom material with additional parameters to database

phoebe-p, 2020-06-01:
If create_new_material is called with only two arguments (name of new material, path to the n and k data files, respectively) everything works fine and the custom_parameters.txt file in the hidden .solcore folder is updated with just a heading corresponding to the relevant materials, data files are copied to a folder in custom_materials and the config file is updated with the relevant path. However, if a file with additional parameters is supplied as a fourth argument, nothing is added to custom_params.txt which means Solcore cannot find the material in the database.

Works fine:
```
create_new_material('SiGeSn', 'SiGeSn_n.txt', 'SiGeSn_k.txt')
material("SiGeSn")()
```

Doesn't work:
```
create_new_material('SiGeSn', 'SiGeSn_n.txt', 'SiGeSn_k.txt', 'SiGeSn_params.txt')
material("SiGeSn")()
```

MichaelNOz, 2020-06-01:
Maybe my Solcore version is behind, but I have been using the equivalent of:
''
create_new_material('SiGeSn', 'SiGeSn_n.txt', 'SiGeSn_k.txt', 'SiGeSn_params.txt')
material("SiGeSn")()
''
However, Solcore does not read the parameters. A workaround for the DA approximation (as an example) was creating the using an instance of the material all the relevant parameters must be supplied:
''
SiGeSn=material("SiGeSn")
p_material = SiGeSn(Na=XXX,electron_diffusion_length=XXX,electron_mobility=XXX,band_gap=XXX,relative_permittivity=XXX, eff_mass_electron=XXX,eff_mass_hh_z=XXX,eff_mass_lh_z=XXX)

''

dalonsoa, 2020-06-02:
It might be the order in which files are read in the create_material function. I’ll have a look ASAP.

BTW, we have brand new issue templates, one specifically for bugs. It will be good to start using them ;)

dalonsoa, 2020-06-02:
I am not able to reproduce the error. Runnig the [`custom_materials_example.py`](https://github.com/qpv-research-group/solcore5/blob/devel/examples/custom_materials_example.py) in a clean `devel` branch works as expected, with all the parameters added to the `custom_params.txt` file. 

Could you check using the `devel` branch?

MichaelNOz, 2020-06-03:
I never had any problems using custom_material_example.py and SiGeSn, only when I tried my own material did it go awry. Below is an example of the files I used. Solcore could load the n and k correctly (for optical simulations), but failed to load the parameters for DA calculations.
However, I haven't explored this fully due to a time constraint.

[In89Ga11As23P77_n.txt](https://github.com/qpv-research-group/solcore5/files/4720599/In89Ga11As23P77_n.txt)
[In89Ga11As23P77_params.txt](https://github.com/qpv-research-group/solcore5/files/4720600/In89Ga11As23P77_params.txt)
[In89Ga11As23P77_k.txt](https://github.com/qpv-research-group/solcore5/files/4720602/In89Ga11As23P77_k.txt)

phoebe-p, 2020-06-15:
Ok, I have just tried again on the current version of the devel branch and it's fine, I think I know what I did before, the heading in the parameters file didn't match the material name so obviously it couldn't be found... silly! Will close this now.

closed 2020-06-15 by phoebe-p