qpv-research-group/solcore5 #125: Radiative saturation current in "diode equation"

AndiPOz, 2020-06-12:
The radiative saturation current in the diode equation module is currently implemented incorrectly. 

The radiative saturation current is an interface property and depends on the smaller of the optical density of states either side of the interface. Therefore, to calculate the radiative saturation current of a junction one needs to know the refractive index of the material above and below as well as the refractive index of the junction itself.

There are essentially four relevant cases for the pre-factors of the integral:

1.) 2 pi/(h^3 c^2) for a junction in radiative contact with air and a back mirror.

2.) (n_i^2+1) 2 pi/(h^3 c^2) for junction i in radiative contact with air and a material (junction or substrate) with equal or higher refractive index.

3.) (n_(i+1)^2+1) 2 pi/(h^3 c^2) for junction i in radiative contact with air and junction or substrate I+1 with refractive index smaller than n_i.

4.) (2 n^_i^2) 2 pi/(h^3 c^2) for junction i in radiative contact with junction above and substrate or junction below with same refractive index.

Note that a bottom junction in a multi-junction stack can still be considered to be in radiative contact with air because all the junctions above are transparent to the emission from the bottom junction. Case 4 arises only if two junctions or more are made from the same absorber (relevant for monochromatic energy conversion).

Because new arguments need to be introduced into the function, this requires care when fixing in order not to break dependent functions.

dalonsoa, 2020-06-12:
Fantastic discusion! 

Just to confirm, you're refering to both:

- the [radiative current](http://docs.solcore.solar/en/master/Solvers/TwoDiode.html#solcore.analytic_solar_cells.diode_equation.calculate_J01) calculation in the 2-diode solver and 
- the correspoinding equation in the detailed balanced solver (given by eq. 30 [in this paper](https://link.springer.com/article/10.1007/s10825-018-1171-3))

right?

AndiPOz, 2020-06-12:
Equation (30) in the paper is correct. But it does not result in the equation written in the radiative current calculation in the 2-diode solver.
Equation (33) gives the clue as to why. The integral there is limited to the critical angle which limits A_front(E) in equation (37) to 1/n^2.
Light trapping does not change this. A lambertian surface only redistributes the internal reflectivity of the surface between angles but cannot reduce the integral over the angle to arbitrarily small values.

dalonsoa, 2021-10-05:
@all-contributors please add @AndiPOz  for bug