qpv-research-group/solcore5 #131: ERROR in SMARTS: [WinError 2] The system cannot find the file specified, ERROR in SMARTS: Output file is empty. Likely error in the input parameters. ERROR in SMARTS: ERROR: SMARTS location not correctly configured or SMARTS executable not working.

mtaddei94, 2020-08-13:
I installed SMARTS on Windows 10 using Ubuntu Shell as described here: http://docs.solcore.solar/en/master/Installation/Solcore_on_Windows.html 
It is correctly working on the command line when typing _./smarts295_ , and produces two text files: “smarts295.ext.txt” and “smarts295.out.txt”.
However, when I try running the example (http://docs.solcore.solar/en/master/Examples/example_light_sources.html) on jupyter notebook it doesn't work and gives the Errors : 

ERROR in SMARTS: ERROR: SMARTS location not correctly configured or SMARTS executable not working.

SMARTS can be obtained free of charge from the NREL webpage:

http://www.nrel.gov/rredc/smarts/

You might need to re-compile the Fortran code as current 64 bit CPUs are not supported by the shipped binaries.

ERROR in SMARTS: [WinError 2] The system cannot find the file specified
ERROR in SMARTS: Output file is empty. Likely error in the input parameters.**To Reproduce**

Steps to reproduce the behavior:

I installed the solcore package for MAC in the Ubuntu folder:
[image]

And SMARTS runs correctly:
[image]

Then when I open jupyter notebook and I try the example you wrote I get those Errors:
 1) First I imported solcore
2) I use _os_ to verify that the path to the folder exists
3) I use config.smarts('path to Ubuntu dedicated folder contaning SMARTS_295_MAC folder)
[image]
 4) I do _print(config)_ to verify that Solcore is looking for smarts in that folder 
[image]

5) I run the example from "http://docs.solcore.solar/en/master/Examples/example_light_sources.html":
[image]

6) I get those ERRORS with SMARTS:

_ERROR in SMARTS: ERROR: SMARTS location not correctly configured or SMARTS executable not working.

SMARTS can be obtained free of charge from the NREL webpage:

http://www.nrel.gov/rredc/smarts/

You might need to re-compile the Fortran code as current 64 bit CPUs are not supported by the shipped binaries.

ERROR in SMARTS: [WinError 2] The system cannot find the file specified
ERROR in SMARTS: Output file is empty. Likely error in the input parameters.
ERROR in SMARTS: ERROR: SMARTS location not correctly configured or SMARTS executable not working_

[image]

dalonsoa, 2020-09-23:
Hi Margherita, 

I think I know what the error is about. In Solcore, you need to configure the path to the folder containing SMARTS. Then, depending on the operative system, Solcore searches for the correct executable within that folder:

- Windows: `smarts295.exe`
- Linux: `smarts295`
- MacOS: `smarts295.command`

You are in Ubuntu - or Windows if you are running python from outside of the Ubuntu shell - but you are using the MacOS version of SMARTS, so Solcore is trying to execute a file (the one corresponding to your OS, either Linux or Windows), that do not exist. The extension is often hidden. You can see it with `ls -al`.

So I would suggest you install the version of SMARTS correspoinding to your OS or change the name of the executable, removing the `.command` at the end. 

I hope this helps!

dalonsoa, 2020-10-01:
@mtaddei94 , did you manage to make this work?

closed 2021-01-13 by dalonsoa

PhilBrk8, 2023-09-07:
I had to recompile the smarts295 executable for a 64-bit system.

for that I had to install the intel fortran compiler first and configure that, which was a pain in the *** itself, hope this overview helps:

Install the "ifx" command ( I´m on fedora ) like this:
https://www.intel.com/content/www/us/en/developer/tools/oneapi/hpc-toolkit-download.html?operatingsystem=linux&distributions=dnf

then configure your environment and run the compiler like this:
https://www.intel.com/content/www/us/en/docs/fortran-compiler/get-started-guide/2023-2/get-started-on-linux.html

----- BUUUUUT:
the setvars script in the description didn´t work properly and I had troubles with shared libraries.

With the help of those two and a whole lot of trial and error...
https://stackoverflow.com/questions/70687930/intel-oneapi-2022-libimf-so-no-such-file-or-directory-during-openmpi-compila
https://github.com/gher-uliege/DIVA/issues/8

...I ended up with the following two lines in my .zshenv file (or put it in your .zshrc, but I like my environment defined in the .zshenv)
source /opt/intel/oneapi/setvars.sh intel64
export LD_LIBRARY_PATH="/opt/intel/oneapi/compiler/2023.2.1/linux/compiler/lib/intel64_lin"