qpv-research-group/solcore5 #139: Create binary wheels for Solcore

dalonsoa, 2020-10-13:
Solcore contains some fortran code - for the Poisson-drfit-diffussion solver - that needs to be compiled and converted into an extension module that can be imported from Python. That is [done automatically when installing Solcore](http://docs.solcore.solar/en/master/Installation/installation.html#installing-solcore) using the `--with-pdd` flag, which employs F2Py under the hood, a tool included in NumPy. 

The caveat is that, for this to work, the user installing Solcore - either from PyPI or directly from GitHub - will need to have a suitable fortran compiler at hand and all the dependencies libraries since this flag disables the use of wheels for any of the Solcore dependencies. All of them will need to be recompiled. It is not a huge deal for Linux and MacOS users, but it is a pain for Windows users. 

The solution is to distribute binary wheels for the three OS, ensuring that [all dependent libraries needed by the fortran code are statically linked with the expansion module](https://stackoverflow.com/questions/44043990/how-to-build-a-python-wheel-with-compiled-fortran-extension-module-without-requi). The solution (for each of the OS) should work in GitHub Actions such that, when there is a new release, the wheels can be automatically uploaded to PyPI. 

This is an umbrella issue that will be updated with the individual, OS specific issues:

- [ ] Linux - Need to use manylinux
- [ ] MacOS
- [ ] Windows

canns99, 2021-07-03:
Sir, I am a beginner and I have some problems with FORTRAN files in Solcore. It took me a long time to deal with them, First, maybe forgive my poor English.
Following the instructions for installing the PDD compiler, an error occurred in the "-- with-PDD" step on the Windows system.
Then I switched to Linux system and continued. That step didn't go wrong, but there were still other problems. In order to use the file "ddModel-Current.f95".
Using the command "python /usr/bin/f2py -m driftdiffusion -c ddmodel-current. f95" Under the Linux terminal , and generate the.so file. Then an error in Python import would appear saying that the module could not be found.
Even through the command "sudo cp-r driftdiffusion. So /usr/lib/python3.6" into the Python library file directory.There is still the "ImporError :.so: undefined Symbol: Pycobject_type" issue.

In short, on how to use "DDmodel - current. f95" documents, namely the solcore "DriftDiffusionUtilities. Py" files "from the DDmodel import driftdiffusion as dd"  this step.
I am really at a loss what to do and I have consulted and sought help in vain. I sincerely hope can get your help

canns99, 2021-07-03:
Sir, I am a beginner. I have some problems with FORTRAN files in Solcore. It took me a long time to deal with them, First, maybe forgive my poor English.
Following the instructions for installing the PDD compiler, an error occurred in the **-- with-PDD** step on the Windows system.
Then I switched to Linux system and continued. That step didn't go wrong, but there were still other problems. In order to use the file "**ddModel-Current.f95**".
Using the command "**python /usr/bin/f2py -m driftdiffusion -c ddmodel-current. f95**" Under the Linux terminal , and generate the.so file.The error is not found in Python import.
Even through the command "**sudo cp-r driftdiffusion. So /usr/lib/python3.6**" into the Python library file directory.There is still the "**ImporError :.so: undefined Symbol: Pycobject_type**" issue.

> In short

Here's my problem, how to use "**DDmodel - current. f95**" documents, namely the solcore "**DriftDiffusionUtilities. Py**" files "**from the DDmodel import driftdiffusion as dd**"  this step.
I am really at a loss what to do and I have consulted and sought help in vain. I sincerely hope can get your help

Abelarm, 2022-10-02:
Hi @dalonsoa, 

do you think this two remaining task are still relevant?

dalonsoa, 2022-10-03:
Hi @Abelarm , most certainly! It's also the case that the linux builds are created only up to Python 3.8, so there's work to do there to extend the support to Python 3.9 and, possibly, 3.10.

Abelarm, 2022-10-03:
> Hi @Abelarm , most certainly! It's also the case that the linux builds are created only up to Python 3.8, so there's work to do there to extend the support to Python 3.9 and, possibly, 3.10.

Perfect! 

i'll probably do two separate PRs since the linux one is already up and running

Abelarm, 2022-10-03:
Hi @dalonsoa I have created two PR:

1. Modified the test for running on python 3.9 and  3.10
2. Added build & deploy for windows 
    - The second PR I fucked up while rebasing the commits so I had to delete it :(

dalonsoa, 2022-10-04:
Hi @Abelarm, This looks great! I've already accepted the linux one.

For the Windows one, have you checked if the binary wheels created do work in a stand-alone, Windows system that do not have a Fortran compiler? I've seen cases where the binary wheels are created but then you still need to install the compiler because some shared libraries are not included in the wheel, which defeats the purpose of having a wheel!

I will try to check that myself later in the week.

Abelarm, 2022-10-04:
> Hi @Abelarm, This looks great! I've already accepted the linux one.
>
>
>
> For the Windows one, have you checked if the binary wheels created do work in a stand-alone, Windows system that do not have a [...]

Actually I wanted to try but I don't know how to download the wheel from the action itself without the publishing of the wheel, and since I use a Linux for dev I can't build it on my own :(

dalonsoa, 2022-10-04:
You will need to work on a branch in your own fork of Solcore with a custom workflow that upload the artefact for download rather than publishing it to GitHub, for example using this action https://github.com/marketplace/actions/upload-a-build-artifact . Another option is to upload it to TestPyPI. From either of those two, you can download it and try the wheel in a fresh system.

But, if you don't have access to a Windows machine, you cannot test it, anyway. Let me try this at some point this week. In principle, the PR looks good and should work, but let's test it before we merge.

Abelarm, 2022-10-04:
I would remove ngspice as you commented, since I never build it for windows I went with there was in the test

dalonsoa, 2022-10-10:
I've run a job that uploads the wheels as artefacts. You can find them here: https://github.com/qpv-research-group/solcore5/actions/runs/3221669724

I'll try the windows ones first thing tomorrow morning, but considering their small size, I'm a bit suspicious about they having all the required dependencies. I'll post my findings.

dalonsoa, 2022-10-11:
Great news! I've just tested the Windows wheels on a fresh Windows 11 system with Python 3.10.7 and they work well without installing anything else! This is going to be a huge leap forward for many people!!

Many thanks for making it happen!

Abelarm, 2022-10-11:
YEAH great news!! Happy to help !!

dalonsoa, 2022-10-27:
Completed in #229 , excluding M1 architecture for Apple

closed 2022-10-27 by dalonsoa