qpv-research-group/solcore5 #141: Set up All contributors bot in Solcore

dalonsoa, 2020-10-13:
Set up the [All Contributors bot](https://allcontributors.org) in Solcore to acknowledge all types of contributions to it, not just code.

dalonsoa, 2021-09-30:
@all-contributors please add @dalonsoa for code, bug, doc, example, ideas, infra, maintenance, review, test

dalonsoa, 2021-09-30:
@all-contributors please add @phoebe-p  for code, bug, doc, example, ideas, maintenance, review, test

dalonsoa, 2021-09-30:
@all-contributors please add @twmwilson for code, bug, doc, example, ideas, review, test

dalonsoa, 2021-09-30:
@all-contributors please add @iclned  for doc, example, ideas, finance review, test

dalonsoa, 2021-09-30:
@all-contributors please add @MarkusFF for code, bug, doc, example, ideas, design

dalonsoa, 2021-09-30:
@all-contributors please add @jcohen02  for  infra

MarkusFF, 2021-09-30:
Thanks Diego!

dalonsoa, 2021-10-05:
@all-contributors please add @PieceMaker for test, review

dalonsoa, 2021-10-05:
@all-contributors please add @mhrownaghi  for test, code

dalonsoa, 2021-10-05:
@all-contributors please add @@federikovi for examples

dalonsoa, 2021-10-05:
@all-contributors please add @federikovi for examples

dalonsoa, 2021-10-05:
@all-contributors please add @Artyko  for code

dalonsoa, 2021-10-05:
@all-contributors please add @emmanuel-carreira for examples

dalonsoa, 2021-10-05:
@all-contributors please add @NimishVerma for examples

dalonsoa, 2021-10-05:
@all-contributors please add @P-Tillmann for code, bug, test

dalonsoa, 2021-10-05:
@all-contributors please add @jmllorens for code, bug, test

dalonsoa, 2021-10-05:
@all-contributors please add @Abelarm for code, infra

dalonsoa, 2021-10-05:
@all-contributors please add @jkrich for code, bug

dalonsoa, 2021-10-05:
@all-contributors please add @MichaelNOz for code

dalonsoa, 2021-10-05:
@all-contributors please add @hrishikeshSuresh for example

dalonsoa, 2021-10-05:
@all-contributors please add @jscooksey for example

dalonsoa, 2021-10-05:
@all-contributors please add @jaibhageria  for example

dalonsoa, 2021-10-05:
@all-contributors please add @yuriy-os for design

dalonsoa, 2021-10-05:
@all-contributors please add @dalonsoa for data

dalonsoa, 2021-10-05:
@all-contributors please add @MarkusFF  for data

closed 2021-10-12 by dalonsoa