qpv-research-group/solcore5 #146: Specified wavelength array overwritten in lightsource when using source_type='standard'

P-Tillmann, 2020-10-23:
**Describe the bug**
When creating an instance of LightSource with source type 'standard' the x_internal property is overwritten in a later step of the instance initiation (in _get_standard_spectrum function). The call to instance.spectrum() does not yield the user specified wavelength but the wavelength of the standart spectrum.

If you consider this behavior a bug, i can submit a PR.

**To Reproduce**
import numpy as np
import solcore
import solcore.light_source as ls

wl_arr = np.linspace(300, 1200, 10)*1e-9

ls_test = ls.LightSource(source_type='standard', version='AM1.5g', x=wl_arr,
                            output_units='photon_flux_per_m')
print(ls_test.spectrum()[0])
[ 280.   280.5  281.  ... 3990.  3995.  4000. ]

**Expected behavior**
print(ls_test.spectrum()[0])
[3.0e-07 4.0e-07 5.0e-07 6.0e-07 7.0e-07 8.0e-07 9.0e-07 1.0e-06 1.1e-06
 1.2e-06]

**Desktop (please complete the following information):**
 - OS: ubuntu 18.04 
 - Solcore Version directly cloned from github (master branch)

dalonsoa, 2020-10-26:
Many thanks for pointing this out. I can reproduce the error and this was certainly not the expected behaviour. Indeed, the x value provided during the creation of the light_source is pretty much ignored and only the one passed when calling `spectrum` or the internal one - coming from a file, for example - are used. 

Please, go ahead with the PR.

dalonsoa, 2020-11-02:
Solved by #147

closed 2020-11-02 by dalonsoa