qpv-research-group/solcore5 #152: SMARTS implementation with Solcore on Linux system.

mdkj96, 2021-01-27:
Hello,
I have just started using Linux for ease of use with Solcore and I am having some trouble implementing SMARTS. I have ran the lightsource.py test file which works well with the SMARTS lines commented out however, gives errors (seen below) when using the smarts functions. I can't seem to find any instruction for installing SMARTS on linux, only for installing in windows, apologies if I have missed something. I will outline the problem below. Any help is much appreciated. 

The steps I have done to get where I am are as follows:

1. I downloaded the SMARTS folder (both mac and linux as I tested both folders with this process)
2. Compiled the file that produces a.out
3. Renamed a.out to smarts295
4. Configured solcore to the folder that contains this executable 
5. Ran the lightsource.py file (http://docs.solcore.solar/en/master/Examples/example_light_sources.html)
6. Received the error (below)

For context line 291 of smarts295.f: OPEN (UNIT=22,FILE='Gases/Abs_O2.dat',STATUS='OLD')

### Error 
```
        ERROR in SMARTS: At line 291 of file smarts295.f (unit = 22)
        Fortran runtime error: Cannot open file 'Gases/Abs_O2.dat': No such file or directory
        
        Error termination. Backtrace:
        #0  0x7f7c6af96d3a
        #1  0x7f7c6af97849
        #2  0x7f7c6af9852f
        #3  0x7f7c6b1d75a6
        #4  0x7f7c6b1d789c
        #5  0x555a7323dffc
        #6  0x555a73259cac
        #7  0x7f7c6ac410b2
        #8  0x555a7323133d
        #9  0xffffffffffffffff
        
        ERROR in SMARTS: Output file is empty. Likely error in the input parameters.
        /home/mjsho/.local/lib/python3.8/site-packages/solcore/light_source/light_source.py:281: RuntimeWarning: divide by zero encountered in true_divide
          2
        /home/mjsho/.local/lib/python3.8/site-packages/solcore/light_source/light_source.py:286: RuntimeWarning: divide by zero encountered in true_divide
          / (np.exp(h * c / (x * kb * T)) - 1)
        /home/mjsho/.local/lib/python3.8/site-packages/solcore/light_source/light_source.py:281: RuntimeWarning: invalid value encountered in true_divide
          2_
```
I hope I have given enough context to this issue.

jannsta1, 2021-02-28:
I have also been trying to get SMARTS (v2.9.5) working with solcore in linux (Ubuntu 20.04) and independently followed similar steps as described in this question and get a similar traceback as above (in my case I don't see the hex block backtrace). @mdkj96 - did you manage to get anywhere with this? If not can anyone provide any help with debugging the issue? In my case the smarts295 application seems to run OK when executed from the command line.

dalonsoa, 2021-03-01:
Hi @mdkj96 , I'm sorry I completely missed this question in January. @jannsta1 , I'll investigate this along the day.

dalonsoa, 2021-03-02:
I have just found the problem. I'm working on a solution and a new version should be ready soon. Many thanks for pointing this out!

jannsta1, 2021-03-02:
Great - thanks for looking at this so quickly. I'm completely new to solcore but just let me know if testing of updated code at my end or anything would be helpful.

mdkj96, 2021-03-02:
@dalonsoa thank you for looking in to this.

dalonsoa, 2021-03-02:
A new version of Solcore has been released [5.7.4](https://pypi.org/project/solcore/) solving this issue. Please, update when you can

PR #154 (2021-03-02): Fix wrong cwd when running SMARTS
The cwd directory was set wrong when running smarts, which affected Linux version but not the others. 

Close #152
closed 2021-03-02 by dalonsoa