qpv-research-group/solcore5 #162: Calculating the reflection, transmission and absorption of a solar cell from external data

Prathamesh-Sahasrabudhe, 2021-06-11:
**Please, confirm the following:**

- [Yes] I have checked [Solcore’s documentation](http://docs.solcore.solar/en/master/) and could not find the answer.
- [Yes] I haven’t found any [other issue](https://github.com/qpv-research-group/solcore5/issues?q=) on a related topic.

**What’s the question?**
Please, be concise and straight to the point

I am trying to find out the optical properties ( reflection, transmission and absorption ) of a simple c-Si solar cell using external optics as the optics method. I want to know how the external data is to be provided in the solar cell definition. 

I want to understand how the functions external_reflected and external_absorbed are defined and implemented. 
Please provide an illustrative example which calculates the i-v characteristics of a simple c-Si solar cell using external optics method.

**Additional information**
Add any other relevant information that could help us answering your question, E.g.:
- Code of a minimum working example showing what you’re trying to do.
- Screenshot showing the problem you are talking about.
- Context in which your question applies

dalonsoa, 2021-06-14:
Hi! Could you describe what exactly you have tried (including the code you have used) and the error messages you are getting?

Prathamesh-Sahasrabudhe, 2021-06-15:
Problem Description:

We have simulated and obtained the RAT (Reflection, Absorption and Transmission) data for a simple c-Si solar cell with SiNx as the anti reflective coating, using the wafer ray tracer on PV Lighthouse. 
We want to pass this data to SOLCORE to obtain the i-v characteristics of the solar cell.
Hence, we are using the  optics method as "external". DA method will be used as the electrical solver (currently not implemented in the code)

The solar cell structure and other parameters have been indicated by appropriate comments in the code. I am also attaching it once again separately. 

Issue Faced:
We are trying to implement the "external_reflected" and the "external_absorbed" functions which are to be provided in the solar cell definition. However we are not able to implement these functions correctly. 

I am attaching the code I have written to implement these functions and the image of the error flagged by SOLCORE.

What I want:
Please let me know how the functions external_reflected and external_absrobed should be written. What is the format and the number of arguments they take. What output is expected of the two functions. 
It will be very helpful if you can provide a simple illustrative example showing how to use external RAT data to obtain the i-v characteristics of the solar cell. 
[externaloptics_final.txt](https://github.com/qpv-research-group/solcore5/files/6652612/externaloptics_final.txt)   --> code file
[siarc.csv](https://github.com/qpv-research-group/solcore5/files/6652613/siarc.csv) --> SiNx anti reflective coating file
[OpticalPropertiesOfSilicon.xlsx](https://github.com/qpv-research-group/solcore5/files/6652621/OpticalPropertiesOfSilicon.xlsx) --> Absorption Depth file obtained from PV Education
[RAT_data_Simple_Si_Solar_Cell.ods](https://github.com/qpv-research-group/solcore5/files/6652637/RAT_data_Simple_Si_Solar_Cell.ods) --> RAT data file obtained from wafer ray tracer on PV Lighthouse
[image] --> Error message

dalonsoa, 2021-06-16:
By the way, I would suggest you have a look at [Rayflare](https://rayflare.readthedocs.io/en/latest/) to calculate the optics. It has been written @phoebe-p, one of the co-creators of Solcore, and it should interface with it pretty well.

Prathamesh-Sahasrabudhe, 2021-07-07:
Dear Dr. Alonso,

I am bringing up another question which is a follow up to the previous one. 
I have incorporated your sugesstions in my code. 
1. I have loaded the data into two text files, one for reflection and one for absorption. 
2. In the functions absorb and reflect I am interpolating the data.
3. I passed the function absorb to the external_absorbed attribute.

After doing these, now I want to plot the I-V curve for this solar cell. The code runs fine and does not give any compilation errors. However, the I-V curve I am obtaining is not correct.

Here are some plausible sources of error I can think of. 
1. I am passing the fractional absorption and not the differential absorption.
2. Is the function absorb behaving the way it is intended to? I am asking this because I have written the function absorb based on my understanding of what is given in the documentation and I am not sure if I have interpreted it correctly. 

I am attaching the data files, the code file, the terminal message and the output again for your reference. Please let me know. 
[WavelengthVsReflection.txt](https://github.com/qpv-research-group/solcore5/files/6779001/WavelengthVsReflection.txt)
[WavelengthVsAbsorptionVsAbsorptionDepth.txt](https://github.com/qpv-research-group/solcore5/files/6779002/WavelengthVsAbsorptionVsAbsorptionDepth.txt)
[ExtOpt.txt](https://github.com/qpv-research-group/solcore5/files/6779007/ExtOpt.txt)
[image]
[image]

phoebe-p, 2021-07-14:
Hi,
I am looking at your file (thanks for providing a nice working example!). I think the absorption data is not in the correct format. I am quite confused by the WavelengthVsAbsorptionVsAbsorptionDepth file - what is the absorption depth in this file? This data should be a 2D array with data for every wavelength and depth in the structure, since differential absorption depends on both wavelength and depth inside the structure (this is also what the function 'absorb' should return; the input is a single depth or a list of depths, and the return is an array with dimensions (number of depth points, number of wavelengths)). I think there is also some issue with units; the position (p in your script) should have units of m, but goes up to '45' even though the width of the cell is 180 um (180e-6 m). 

Is the experimental data you have just wavelength vs. total absorption? If so, you need to calculate the differential absorption depending on the depth using the Beer-Lambert law or another approach that suits the structure.

I will get on with writing an example showing what the absorb function should look like, and how it can be calculated (approximately) from total absorption data for a thick cell like this, since we don't currently have an example like this.

Prathamesh-Sahasrabudhe, 2021-07-14:
Hello Dr. Phoebe,

I am attaching a file "Optical Properties of Silicon" from where I am taking the necessary data in my "WavelengthVsAbsorptionVsAbsorptionDepth" file. In it you will get the fractional absorption, absorption depth at each wavelength.  The units used in the WavelengthVsAbsorptionVsAbsorptionDepth.txt file for the absorption depth (i.e. the third column) are in "cm" and not "m" I am sorry for that. 

We are considering wavelengths from 300-1200 nm.  This is because a 180 um thick si solar cell absorbs mostly upto 1000 nm but after modifying the front and rear surface morphology of si solar cell, we can improve the absorption upto 1200 nm by increasing the pathlength of light.

In response to your second paragraph, the data is fractional absorption at each wavelength and hence we will have to calculate the differential absorption separately. 

Thank you for clarifying what the function "absorb" should do. As you have mentioned, please do write an example for this as it will greatly help to gain a perspective. 
[OpticalPropertiesOfSilicon.ods](https://github.com/qpv-research-group/solcore5/files/6816266/OpticalPropertiesOfSilicon.ods)

phoebe-p, 2021-07-28:
See attached an example of how you can calculate an absorption profile (differential absorption) from total absorption data, and how to put this in the correct format for Solcore. I will also add this to the examples in the repository to help others in future, I hope it clarifies things!
[external_optics.txt](https://github.com/qpv-research-group/solcore5/files/6893266/external_optics.txt)

closed 2022-03-18 by phoebe-p