qpv-research-group/solcore5 #163: E0 calculations don't quite match up to literature

warlord385, 2021-06-19:
When calculating the E0 of "Ga0.49In0.51P"  using solcore.get_parameter("Ga0.49In0.51P", parameter="E0", T=300) I get 2.0311.  This is a bit far off from using the formula 1.34 + 0.511x + 0.6043x^2 which yields 1.735.

dalonsoa, 2021-06-21:
I indeed can confirm this. When getting `E0` step by step, things seems to suggest it is not applying the bowing:

```python
E0GaP = solcore.get_parameter("GaP", parameter="E0", T=300)
E0InP = solcore.get_parameter("InP", parameter="E0", T=300)

x = 0.51
E0_NoBow = GaP*(1-x) + InP*x
E0_bow = GaP*(1-x) + InP*x - x*(1-x)*0.65   # This is the bowing according to Vurgaftman APR 2001
```

This shows that `E0_NoBow = 2.031 eV` and `E0_bow = 1.87 eV`, as you indicated. 

I'll investigate this issues, as it seems a regression - Eg of ternaries was calculated OK in the past. Many thanks for pointing this out!

dalonsoa, 2021-06-22:
I've just realised that this is not a bug, but actually the expected behaviour. My reasoning above was mistaken. 

`E0` is one of the Adachi's critical points, as described in his papers (see below), and there is no bowing information in those papers helping to provide a more accurate value for `E0` in the case of `GaInP`.

If you are looking for the band gap, you should ask for the parameter `band_gap`, which will give you the lowest energy transition, or for `Eg_Gamma`, `Eg_X` or `Eg_L` which will give you the energy transition at the chosen point. These parameters are taken from Vurgaftman, who does provide bowing information. 

I hope this helps!

1. Adachi, S.: Model dielectric constants of GaP, GaAs, GaSb, InP, InAs, and InSb. Phys. Rev. B 35(14), 7454–7463 (1987)
2. Adachi, S.: Optical dispersion relations for GaP, GaAs, GaSb, InP, InAs, InSb, Alx Ga1−x As, and In1−x Gax Asy P1−y . J. Appl. Phys. 66(12), 6030–6040 (1989)
3. Vurgaftman, I., Meyer, J.R., Ram-Mohan, L.R.: Band parameters for III–V compound semiconductors and their alloys. J. Appl. Phys. 89(11), 5815–5875 (2001)

closed 2021-08-24 by dalonsoa