qpv-research-group/solcore5 #164: Running the MJ Solar Cell Tutorial.py program has some problems

canns99, 2021-07-17:
After the **MJ Solar Cell Tutorial.py** program runs，figure shows the summary of the efficiency, Isc, Voc and fill factor (FF) versus concentration but goes wrong.
I didn't change any code, but as the figure shows, the result is completely incorrect.
[image]

•The correct answer would be
[image]

Is there a problem with the FORTRAN file or something?Any help would be greatly appreciated

phoebe-p, 2021-07-29:
Indeed, this is the wrong result -- I am looking into it to see if I can find where the problem is.
Just a note on the running time: the MJ_solar_cell_tutorial.py does say in a comment:

```
# NOTE: Due to the presence of QWs and the fact we calculate things a 19 different
# concentrations, this might take a while (~4 hours).
# Remove the QWs as indicated above to test the code much faster.
```

If you comment out the `+ QW_list` line in the solar cell definition (so not including the quantum wells), as indicated, this will run much faster. Of course you can also reduce the number of concentrations to reduce the runtime.

phoebe-p, 2021-07-29:
The issue seems to be with this line in the for loop:

```
light_source.options["concentration"] = c
```
As it turns out, this does not actually change the concentration in the calculations. Changing it to:

```
light_source.concentration = c
```
gives the expected results. @canns99 thank you for flagging this, obviously we will update it in the example! @dalonsoa do you think this is due to a recent change in light_source? As far as I can tell the light_source.options construction is used only in MJ_solar_cell_tutorial.py and it's ipynb equivalent, no other examples.

dalonsoa, 2021-07-29:
Many thanks for pointing this out. As that tutorial is not run with the test suite – and this one is quite incomplete, anyway – we missed that inconsistency after changing the way `light_source` was working, as @phoebe suggest. We will update the tutorial .py and ipybnb files and the documentation as soon as possible to correctly reflect that change.

dalonsoa, 2021-10-05:
@all-contributors please add @canns99  for bug

closed 2023-02-17 by phoebe-p