qpv-research-group/solcore5 #170: Using S4 with Solcore on python 3

TWDavid1, 2021-09-30:
When I try to run one of the files (“Comparison of Optical Models”), I get an error message “'S4' has no attribute 'New'”. Is there a way to solve this issue since I am using python 3. I have attached the code for reference.
[Comparison_of_Optical_Models.txt](https://github.com/qpv-research-group/solcore5/files/7258815/Comparison_of_Optical_Models.txt)

phoebe-p, 2021-10-12:
Hi @TWDavid1, I need more info to be able to help you: what operating system are you using? Did you install S4 ([instructions here](https://github.com/phoebe-p/S4)? If yes, did the installation seem to work or did it give an error?

phoebe-p, 2021-10-12:
Also, did you clone from the base S4 repository or from https://github.com/phoebe-p/S4? The instructions there won't work if you cloned from https://github.com/victorliu/S4

TWDavid1, 2021-10-12:
Thankyou, it is now working. I had cloned S4 from a different repository so when I was cloning from the correct version (https://github.com/phoebe-p/S4) it was not overwriting it. After deleting the old S4 folder and cloning from https://github.com/phoebe-p/S4 everything seems to be working correctly.

Thanks

closed 2021-11-07 by dalonsoa