qpv-research-group/solcore5 #205: Semi-analytic solution for DA solver using Green's function method

vathomass, 2021-11-02:
Hi,

#### Background:
The depletion approximation (DA) module currently employs the scipy boundary value numerical solver (solve_bvp) to determine the profile of the carrier concentration inside a given junction. Note that for calculating the QE and IV curves we are only interested in the derivative of the carrier concentration at a single point to compute the junction current density.

#### Proposal:
The differential equation in question is linear and a semi-analytic solution can be found using the Green's function method (see this [wikipedia article](https://en.wikipedia.org/wiki/Green%27s_function) for example). This allows to perform the calculation at a single point, substituting the boundary value problem with a numerical integration. A detailed derivation of the solution for the carrier concentration using the Green's function method can be found [here, in appendix A](https://doi.org/10.1088/1748-3190/ac095b). The mathematical model behind the DA module (the boundary value problem itself and the accompanying boundary conditions) remains totally unaffected, only the solution method changes.

#### Benefits:
* Green's function method is not an approximation, but an exact (in theory) solution. The numerical approximation of the integral does introduce errors though.
* A significant speedup is expected.
* The main implementation (not accounting for selecting between bvp or Green's method) is few lines of code.
* No extra dependencies are introduced into the code.

#### Experiment:
To quantify the 2nd point from above, I added the code to compute the solution using the Green's method into the DA solver. I can switch between the two implementation (scipy's solve_bvp and Green's method) by commenting and uncommenting specific lines of code manually. As a benchmark I used the [example "3J with the DA solver"](http://docs.solcore.solar/en/master/Examples/example_3J_with_DA_solver.html) (btw the title in the documentation seems off, it reads "2J cell with the PDD solver"). The attached figures show the comparison between the results obtained using the two methods, which in my opinion are indistinguishable.
[image]
[image]

For the shake of completeness, I computed the infinity norm (maximum absolute point-wise difference) between the curves obtained from the two methods and found to be 0.002 for the QE curves and 0.133 for the IV curves.
The speedup for the QE computation for the cell including 3 junction was measured from 7.8 sec with the solve_bvp to 0.94 secs with the Green's method. The computation time for the IV curve was measured at 0.12 secs and 0.10 secs for the two methods respectively (measurement using the python datetime module).

Would you be interested in adding this solution method in Solcore5?

Best Regards,
Thomas

dalonsoa, 2021-11-04:
Hi Thomas, 

Thank you very much for taking the time to look into this. Adding the Green Functions method to the available options in Solcore has been a long standing desire for us, but never found the time - nor expertise - for doing it. So, absolutely, it will be really good to have your contribution incorporated into Solcore. 

Given that you have some code already running and comparing with the existing results, I gather that much of the work is already done, but let us know if you need guidance or support of where exactly the changes should be made. Maybe the simplest thing to kick this off would be for you to open a draft PR and work on this from there. 

Best wishes, 
Diego

vathomass, 2021-11-07:
Hi Diego and Jose,

Thank you very for the comments.

I will proceed as Diego suggests, I will fork the project and make a draft pull request as soon as I have something concrete for you to review. For sure at some point I will need some guidance on which is the best way to implement the switching between different solving methods, but we can leave this for later.

I hope to have something probably in one week or two.

Best Regards,
Thomas

closed 2022-01-04 by vathomass