qpv-research-group/solcore5 #208: REGARDING QUASI 3D SOLVER IN SOLCORE

Sagnikbas, 2021-11-22:
**Please, confirm the following:**

- [ ] I have checked [Solcore’s documentation](http://docs.solcore.solar/en/master/) and could not find the answer.
- [ ] I haven’t found any [other issue](https://github.com/qpv-research-group/solcore5/issues?q=) on a related topic.

**What’s the question?**
Please, be concise and straight to the point

**Additional information**
Add any other relevant information that could help us answering your question, E.g.:
- Code of a minimum working example showing what you’re trying to do.
- Screenshot showing the problem you are talking about.
- Context in which your question applies
plt.imshow(Vall[:, :, -2, -1])

What does the indices -2 and -1 mean in the Quasi 3D code in solcore documentation(examples) page ?

dalonsoa, 2021-11-26:
This needs to be properly described in the documentation, but there is some information in the `solve_quasi_3D` docstring - in [here](https://github.com/qpv-research-group/solcore5/blob/2ca5ffb3a34e076023ce7604259db07945880c0a/solcore/spice/quasi_3D_solver.py#L27). 

- The `-1` means that we are asking for the last voltage step - in the case of the example, 1.3 V. 
- The `-2` means that we are asking foir the voltage of the top junction, just under the top series resistance.

As I said, this should be explained in the documentation, so thanks for pointint it out.

Sagnikbas, 2021-11-29:
Dear Sir, 
 We are using crystalline Silicon solar cell 1 junction, and using quasi 3D solver to generate the voltage map. The junction definition which I am using is 
db_junction2 = Junction(kind='2D', T=T, Tref=298, j01=702e-11, j02=25.9e-5,Eg=1.12,A=1,jsc=372, R_sheet_top=80, R_sheet_bot=0,
                       R_shunt=1.6, n=3.42), is this description correct?
jo1,jo2,jsc  values are all measured ones and operating at Vmpp = [[0.53V]]([image])
I am attaching the voltage map, the metal grids having voltage 0.53V, while other regions have higher voltage,
[image]
please look at it.

Sagnikbas, 2021-12-06:
Respected Sir,
  Can you send me a sample code regarding Quasi-3D Model to generate
voltage map for Indirect Bandgap Material like Silicon?
Here we are inserting measured values of Jo1,Jo2 and Jsc, so in the 2Diode
model code , I think the code is not considering the light source i.e
"Standard 1.5G " source, so the generated voltage map and IV curve for
quarter mask and the entire mask is not coming same.
If you can send a 2 Diode model code for Si solar cell, considering the
1.5G light source will be really helpful.

Yours sincerely
Sagnik Basu
Indian Institute of Technology Madras