qpv-research-group/solcore5 #209: PDD installation fails on Apple M1 architecture

phoebe-p, 2022-01-27:
**Describe the bug**
Installing Solcore with the PDD fails on new-generation MacBook (Apple M1 Max chip). This appears to be due to the `-march=native` flag used during compilation of the Fortran code.

**To Reproduce**
Steps to reproduce the behavior:

```
pip install solcore
pip install --no-deps --force-reinstall --install-option="--with_pdd" solcore
```

**Expected behavior**
Expected Solcore to install with the PDD solver.

**Result**
A long error message is produced, but the key issue appears to be the following:

```
 CCompilerOpt.dist_test[581] : CCompilerOpt._dist_test_spawn[716] : Command (clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -I/Users/phoebe/Documents/develop/mainenv/include -I/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c /Users/phoebe/Documents/develop/mainenv/lib/python3.9/site-packages/numpy/distutils/checks/test_flags.c -o /var/folders/60/mjbn28rj1ql_r4w004g_ryt00000gq/T/tmp7pyszm34/Users/phoebe/Documents/develop/mainenv/lib/python3.9/site-packages/numpy/distutils/checks/test_flags.o -MMD -MF /var/folders/60/mjbn28rj1ql_r4w004g_ryt00000gq/T/tmp7pyszm34/Users/phoebe/Documents/develop/mainenv/lib/python3.9/site-packages/numpy/distutils/checks/test_flags.o.d -march=native) failed with exit status 1 output ->
    clang-11: error: the clang compiler does not support '-march=native'
```

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Solcore Version [e.g. 22]

**Additional context**
This problem has been [noted elsewhere](https://stackoverflow.com/questions/65966969/why-does-march-native-not-work-on-apple-m1) and also occurs when building [S4](https://github.com/phoebe-p/S4). In that case, it was fixed by replacing the `-march=native` flag by `mcpu=apple-m1`, but I am not sure where the march=native flag even comes from in the case of Solcore.

phoebe-p, 2022-02-21:
@dalonsoa I think I might be able to solve this, but I am not that familiar with how the PDD solver is compiled and just searching the codebase indicates that `march=native` is not set manually anywhere - in setup.py the only relevant thing I can see is that extra_link_args is only set for compiling on Windows. Do you have any thoughts on where to start with this?

dalonsoa, 2022-02-22:
Thanks for pointing this issue and sorry for not being too responsive lately. 

The compilation options are set here:

https://github.com/qpv-research-group/solcore5/blob/883a31abeaaed9bd16c2acab6185e7dd5b217e90/setup.py#L46-L53

That `-march=native` is not set there, so I can only assume that it is the default used by `f2py`. I would suggest you try to overwrite it by explicitly providing the value you want there (or at least one that doesn't cause any trouble) and add the one you need - possibly taking into account the differences in platform. I guess it should be set as `extra_f90_compile_args` in the `Extension` call. More info about this - but not much, to be honest - in [here](https://numpy.org/devdocs/reference/generated/numpy.distutils.core.Extension.html).

phoebe-p, 2022-03-18:
Update (for myself): there appear to be [a more general issue](https://developer.apple.com/forums/thread/651476) with compiling Fortran on the new Apple silicon chips. Currently trying to see if one of these [experimental releases](https://github.com/fxcoudert/gfortran-for-macOS/releases) will work.

phoebe-p, 2022-03-18:
Ok, sort of resolved. For reasons I do not understand, Homebrew gfortran (part of gcc, version 12.0.0 20211218 Target: aarch64-apple-darwin21) completely fails to compile the Fortran source files. It gives a bunch of errors with a similar format like:

```
DDmodel-current.f95:6:17:
Error: Kind 16 not supported for type REAL at (1)

DDmodel-current.f95:1386:20:
Error: Symbol 'sum3' at (1) has no IMPLICIT type
```

This happens whether I try to run the compilation through setup.py, or by calling f2py directly (this is a very small selection from probably hundreds of similar errors). However, downloading an experimental release for ARM macOS 12 (Monterey) from the link in my previous comment, specifically the 19 December 2021 release, did work. However, I am not sure how to specify the path of gfortran manually for f2py, so to get this to work (after installing the experimental gfortran release - follow the instructions given in the README file provided with the download, you need to install the Xcode app and command line tools first), I had to do:

```
brew uninstall --ignore-dependencies gfortran         
pip install -e ".[dev]"
pip install -e ".[dev]" --install-option="--with_pdd" 
brew install gfortran
```

After this, I was able to run examples involving the PDD, however they DID NOT work if trying to import anything involving the PDD when the working directory is the Solcore directory (in fact this is an issue I have had before and have helped others with, we should add it to the documentation). 

Obviously this is annoying because it means having to download Xcode (which is like 10GB) and a specific gfortran only to install the PDD. I have no idea if this will get resolved with future gcc/gfortran versions available through a simpler channel like Homebrew. Hopefully I can at least modify setup.py so it selects the correct gfortran if an Apple silicon/ARM chip is detected (but again, I don't currently know how to do this).

phoebe-p, 2022-10-18:
Update: this issue appears to be fixed using current Homebrew gcc/gfortran (gcc 12.2.0). Documentation should be updated to reflect this.

closed 2023-02-14 by phoebe-p