qpv-research-group/solcore5 #215: Wheels for the Linux and the i686 architecture fail to build

dalonsoa, 2022-04-07:
The binary wheels for the Linux and the i686 architecture fail to build. As a result, none of the Linux wheels (the i86_64 are built ok) are uploaded to PyPI. I feel that i686 is not really a common use case anymore and can safely skip it. 

@jcohen02 , since you implemented this and are more familiar with Manylinux, could you have a look at this? If it looks as if it will take  more than 5 min to fix, just remove the support for i686, please.

jcohen02, 2022-04-07:
Happy to take a look and remove i686 support, will aim to sort this out sometime today.

jcohen02, 2022-04-12:
Apologies for the delay, this should be addressed in PR #216, see the PR for further details.

PR #216 (2022-04-28): Update the manylinux build config for package build/deploy
This resolves the problem highlighted in #215.

The build of the i686 manylinux packages started to fail causing the deployment workflow to fail.

The long string of errors appeared to have something to do with AVX512. I noted some changes in the [release notes for numpy 1.22](https://numpy.org/doc/stable/release/1.22.0-notes.html) relating to AVX512 and its use in vectorizing the `umath` module. I assume this issue is related to the use of numpy 1.22 in an i686 environment and, for the i686 build, I therefore set it to use numpy v1.21 [... cut]
closed 2022-04-28 by dalonsoa