qpv-research-group/solcore5 #220: Error when installing PDD solver

Chaz-Photonicity, 2022-09-28:
**Describe the bug**
I think the following describes the problem
  × Running setup.py install for solcore did not run successfully.
  │ exit code: 1
  ╰─> [496 lines of output]
      /tmp/pip-install-wbzbkh_e/solcore_df0ba3f62bc941f3a99e69aacb9d307e/setup.py:14: DeprecationWarning:
      
        `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
        of the deprecation of `distutils` itself. It will be removed for
        Python >= 3.12. For older Python versions it will remain present.
        It is recommended to use `setuptools < 60.0` for those Python versions.
        For more details, see:
          https://numpy.org/devdocs/reference/distutils_status_migration.html

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
Bug free install.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Microsoft Windows [Version 10.0.19044.2006]
 - Solcore Version: don't know how to find this.  Since i'm just install solcore now .. would be the most updated one?
 - Python 3.10.6

**Additional context**
Prior to running 
~$ pip install solcore --no-deps --force-reinstall --install-option="--with_pdd" solcore
I had performed the following
~$ sudo apt -y install gfortran-11
~$ pip install solcore

phoebe-p, 2022-09-28:
Thank you for reporting this issue, haven't seen this before - do you have a computer with an Arm chip?

phoebe-p, 2022-09-28:
@dalonsoa from the log I think the issue may be similar to what is happening on new Macs with M1 chips, since this seems to be an Arm chip PC rather than Intel/x86.

dalonsoa, 2022-09-28:
Mmmm... interesting. Do you know the solution in that case?

Chaz-Photonicity, 2022-09-28:
FYI: I have INTEL chip set

[image]

Chaz-Photonicity, 2022-09-28:
Maybe just my luck ... got a similar error on my Macbook too :(
Here's my log: https://docs.google.com/document/d/1byGtFhir_azQctR2vCnNk_UNVMnNtTI_LmLWMFfQT0c/edit?usp=sharing

Chaz-Photonicity, 2022-09-28:
> I understand that while you are running Windows, you're installing Solcore in the WLS, right?
> 
> In theory, Solcore has binary wheels for linux so installing the `PDD` manually should not be needed, but I've just noticed that the last version - 5.7.7 - do not have those wheels. Moreover, the [wheels for Solcore 5.7.5 only include Python 3.7 and 3.8](https://pypi.org/project/solcore/5.7.5/#files), which is obviously no good for your case.
> 
> I'll investigate why the binary wheels for 5.7.7 did not build and also try to expand the supported python versions. As long as the dependencies support them, there is no reason for Solcore not to do the same.
> 
> Nevertheless, compilation from source should work. It seems from the log that it is that it is not finding the fortran compiler. Have you check if `gfortran` exists and is in your path? What happens if you just install `gfortran` without specifying its version?
> 
> BTW, you have put together an amazing log about your journey for installing Solcore. It will be useful to exploit that work you have done in some sort of Troubleshooting that other potential Solcore users can follow.

Checking PATH before  install `gfortran` without specifying its version
[image]

phoebe-p, 2022-09-28:
Ah thanks for checking @Chaz-Photonicity - looking again I can see the log is correctly identifying x64 architecture, I just saw something in the error about “ArmFlang” but didn’t look carefully enough. In that case it is strange this doesn’t work. I don’t have any immediate ideas but I can have a look at it next week unless you have some immediate thoughts @dalonsoa (apart from installing gfortran without specifying the version as you mentioned above)

Chaz-Photonicity, 2022-09-28:
> Ah thanks for checking @Chaz-Photonicity - looking again I can see the log is correctly identifying x64 architecture, I just saw something in the error about “ArmFlang” but didn’t look carefully [...]

Removed gfortran-11
and 
sudo apt install gfortran
--> i think it's good now.  but I still have the "import" problem as detailed in #221 

Output: 
[image]

Chaz-Photonicity, 2022-09-28:
just checked the gfortran version number.  I think it's same as the one I'd removed.
[image]

phoebe-p, 2023-02-17:
See #221 and #222!

closed 2023-02-17 by phoebe-p