qpv-research-group/solcore5 #221: Error upon running $ import solcore

Chaz-Photonicity, 2022-09-28:
**Describe the bug**
Upon running $ import solcore
import import: Unable to open XServer () [No such file or directory].

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
-OS: Microsoft Windows [Version 10.0.19044.2006]
-Solcore Version: 5.7.7
-Python 3.10.6

**Additional context**
Prior to running to this error

~$ sudo apt -y install gfortran-11
~$ pip install solcore
~$ pip install solcore --no-deps --force-reinstall --install-option="--with_pdd" solcore

As documented in issue #220, there was a problem with installing PDD.

Chaz-Photonicity, 2022-09-29:
FYI: I've tried these commands with no avail: 

#!/usr/bin/env python
 
#!/usr/bin/env python3

#!/usr/bin/python

dalonsoa, 2022-09-29:
Try entering into a python shell first and then import Solcore. Solcore is a python package, not a shell tool, so you must be within a python interpreter - or a python script - in order to use it. You should see something like the following:

```bash
bash-5.0$ python
Python 3.9.7 (default, Sep 16 2021, 08:50:36)
[Clang 10.0.0 ] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import solcore
WARNING: The RCWA solver will not be available because an S4 installation has not been found.
>>>
```

phoebe-p, 2022-09-29:
Hi, as Diego said - Solcore is a Python package and you are trying to run it from the command line. So either type `python` or `python3` in the command line, or (probably more useful) open it in the IDE you are using (I think VSCode?), this might help: https://code.visualstudio.com/docs/python/python-tutorial

Chaz-Photonicity, 2022-09-29:
Thank you, Diego and Pheobe!this  seems to be going if I would follow  Diego's
recommendation.I have no prior experience in Python ... I followed all the
linked instructions.  Still it was not obvious to me that I had to be in a
"python shell".I did later try the following without luck:#!/usr/bin/env python
But i suppose  you would only use this line in a set of code written in a *.py
file.  Anyway .. i'm  still very new to Python.  Thank you so much for dragging
me along.
NOw I have this other problem on my Mac

Best,Charles Ho
Sent with Mixmax

Chaz-Photonicity, 2022-09-29:
As i was  saying, NOw I have this other problem on my Mac as documented in
#222.  I would like to get the PDD going there too to speed things up.

THanks again.  

Best,Charles Ho

Chaz-Photonicity, 2022-09-29:
Maybe i should  reinstall brew?  

Best,Charles Ho  

Best,Charles Ho

Chaz-Photonicity, 2022-09-30:
Alight moving along ... got things to run.Trying the DB PCE map on VS Code. 
done about can't find the plotsso .. went back to the Python tutorial Phoebe had
recommended. Followed the procedures there.  still no luck.SOrry to pest  you
guys again.  but very frustrating.
Thank you so much!!

phoebe-p, 2022-09-30:
I don’t use VSCode so I’m not sure - just because you were trying different things, could you clarify which environment you’re now in (MacOS or Windows + Linux subsystem) so it’s easier to help? Also, just for our information, is the PDD also working now, and if so, do you know how you solved the issues you were having previously?

Chaz-Photonicity, 2022-09-30:
sorry for confusion .. but yea . i'm pretty desperate to get things moving .. so
try to bring online as much resource as i can to hopefully get something going. 
 hope  you can understand.  So concurrently working on both.  To be honest,
judging from time  it takes for running some of these examples ... will likely
keep working on both laptops going forward.
so the respective status is largely recorded in my log.  but to summarize1) PC:
PDD ok now.  import  problem also got over that.  now  can't plot.2) Mac: PDD is
not still not installed properly.  import OK.  can display plots.
I try to avoid cutting out a partition for LINUX on my windows/PC just to do
solcore... but if you think this maybe  the easier way forward ... i guess  i
will  just do it.  

Best,Charles Ho

phoebe-p, 2022-09-30:
Do any of the things discussed in this issue help? https://github.com/microsoft/vscode-remote-release/issues/452

Beyond that, I am afraid I can’t help - this seems to be an issue with VSCode/WSL rather than Solcore, and it is not really feasible for us to check every combination of operating system/programming environment to give detailed help… 

Re: the other issues you are still having with the PDD, I will keep looking into this on Mac. Which examples are you running which are particularly slow?

Chaz-Photonicity, 2022-09-30:
Totally can understand .. i was  just hoping you might have heard about what i'm 
 going thru  with this VSCode/WSL problem or have a way for me to diagnose what
the issue can be.  Will check your link later.
As with the Mac, since PDD is not done right .. i haven't  tried very many
examples.  I have only tried the TJ example.  Let me know if I should try more
examples to give you more information to work with.  

microsoft/vscode-remote-release#452

Best,Charles Ho

Chaz-Photonicity, 2022-09-30:
Thank you so much again!

Regarding my VSCode/WSL problem ... is there a good place/forum where i can 
find help?

phoebe-p, 2022-09-30:
Well, the link I sent is to the VSCode GitHub, so I suppose they may be able to help you there if that issue does not address the issue. If you are concerned about speed, I would say a Ubuntu (Linux) partition is certainly the way to go. You will never get the same performance running a virtual machine/subsystem inside Windows.

Chaz-Photonicity, 2022-09-30:
Any idea how much faster a dedicated partition would be?  

Best,Charles Ho

phoebe-p, 2023-02-17:
As with the other issue I just closed #222, installation on Windows should also be easy now. If not, feel free to re-open!

closed 2023-02-17 by phoebe-p