qpv-research-group/solcore5 #222: Error when installing PDD on Mac

Chaz-Photonicity, 2022-09-29:
Describe the bug
I think the following describes the problem
While installing PDD solver: pip3 install --no-deps --force-reinstall --install-option="--with_pdd" solcore
>  Running setup.py install for solcore ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for solcore did not run successfully.
  │ exit code: 1
  ╰─> [482 lines of output]
      /private/var/folders/fb/jwjj7_rx2cg5zv2d2hf8d58r0000gq/T/pip-install-6i_4fint/solcore_8039462e4fc4489dbd91a7b502af73c5/setup.py:14: DeprecationWarning:
      
        `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
        of the deprecation of `distutils` itself. It will be removed for
        Python >= 3.12. For older Python versions it will remain present.
        It is recommended to use `setuptools < 60.0` for those Python versions.
        For more details, see:
          https://numpy.org/devdocs/reference/distutils_status_migration.html

If it helps here my log for installing process:
https://docs.google.com/document/d/1byGtFhir_azQctR2vCnNk_UNVMnNtTI_LmLWMFfQT0c/edit?usp=sharing

To Reproduce
Steps to reproduce the behavior:

Expected behavior
Bug free install.

Screenshots
If applicable, add screenshots to help explain your problem.

Desktop (please complete the following information):

OS: Catalina 10.15.7
Solcore Version: 5.7.7
Python 3.10.7
GNU Fortran (Homebrew GCC 12.2.0) 12.2.0
Processor: INTEL

Additional context
Prior to running
~$ pip3 install solcore --no-deps --force-reinstall --install-option="--with_pdd" solcore
I had performed the following
Installed Homebrew GCC
~$ pip3 install solcore
Seems both were installed successfully as far as i can tell.

dalonsoa, 2022-09-29:
The log file for the error in the MacOS case is not complete so it is not clear what's going wrong. Could you include it in full?

Chaz-Photonicity, 2022-09-30:
Updated.  thanks  

Best,Charles Ho

phoebe-p, 2022-09-30:
I cannot reproduce this issue on Mac in Python3.10, but I am on MacOS Monterey (12.6). Looking at your log, it seems like to compile the PDD, your computer is not finding the Homebrew version of gcc/gfortran (from running `brew install gcc`), but it is instead using this one:

```
INFO: Found executable /usr/local/bin/gfortran
```

For comparison, the equivalent line from my installation log looks like this:

```
INFO: Found executable /opt/homebrew/bin/gfortran
```

I am not sure if this gfortran gets installed when downloading XCode (XCode is not actually necessary for installing solcore, as far as I know). 

I do not know if it would change anything, but my Python installation is from Homebrew (`brew install python@3.10`) rather than by downloading from the python.org website.

Chaz-Photonicity, 2022-09-30:
um ... I'm afraid even if i try  to uninstall xcode/python .. it may not change
how it's looking for gfortran.  Is there a place .. maybe in the codes i can
specify where to look for gfortran?  

Homebrew (brew install ***@***.***) rather than by downloading from the
python.org website.

Best,Charles Ho

Chaz-Photonicity, 2022-10-11:
Hi Phoebe and Diego,  hope you have been well.
So I had tried  some more things .... as far as my Mac woes go.
You may check detail  in my log:In short, I'd checked if I had more than one
version of gfortran.Here's what i got  from my finder: see attached screenshots.
At /usr/local/bin/gfortran, there is just an alias pointing to the gfortran-12
installed by homebrew./usr/local/Cellar/gcc/12.2.0/bin/gfortran-12
Firstly, I tried using the update-alternatives command to ensure the system
would look into the following path for gfortran.No luck.Then I'd tried  removing
and reinstalling HOMEBREW.No luck either.
I have started to think maybe Python knows where to find gfortran based on the
following commands.I mean could it be something else other than the gfortran?
Catherines-MacBook:~ HuniCharlie$ gfortran --version

GNU Fortran (Homebrew GCC 12.2.0) 12.2.0

Copyright (C) 2022 Free Software Foundation, Inc.

This is free software; see the source for copying conditions.  There is NO

warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Catherines-MacBook:~ HuniCharlie$ which gfortran

/usr/local/bin/gfortran

Catherines-MacBook:~ HuniCharlie$ which gfortran-12

Catherines-MacBook:~ HuniCharlie$

um .....

Anyway ,have a nice wkend.

Best,

Charles

Best,Charles Ho  

Best,Charles Ho

jcohen02, 2022-10-12:
Hi @Chaz-Photonicity, thought I'd provide some feedback on this since I have access to an Intel-based Mac running 10.15.7  and I thought I'd see if I could reproduce the problem you're experiencing.

I can't reproduce the exact issue and, by default, the installation works fine for me but hopefully my experience may provide some further information. 

The system I tested on is using MacPorts rather than Homebrew (although that shouldn't matter), it's running Python 3.10 installed via MacPorts. The gfortran installation is from the binary releases at [https://github.com/fxcoudert/gfortran-for-macOS/releases](https://github.com/fxcoudert/gfortran-for-macOS/releases) (as linked from [GFortranBinariesMacOS](https://gcc.gnu.org/wiki/GFortranBinariesMacOS)) which installs into `/usr/local/gfortran` with a symlink for the main binary into `/usr/local/bin`, so `which gfortran` returns `/usr/local/bin/gfortran`. Initially the system had an older release of gfortran - 8.2.0.

I tested the build in a clean Python virtual environment. So, running something like:

```
$ mkdir solcore-test; cd solcore-test
$ virtualenv env
$ source ./env/bin/activate
$ pip3 install -v --no-deps --force-reinstall --install-option="--with_pdd" solcore
```

The install completes successfully including the build of the native PDD library. I did notice a small number of errors in the output of the build process but I think this may be related to testing for different CPU features and the build ultimately reports as successful and the native library is built.

Wondering if, perhaps, the problem is with newer versions of gfortran, I updated from 8.2.0 to 10.2. Everything still works as expected.

I also tried removing the versions of gfortran installed from the binaries at [https://github.com/fxcoudert/gfortran-for-macOS/releases](https://github.com/fxcoudert/gfortran-for-macOS/releases) and the system then falls back to a version of gfortran 10.3.0 installed via macports. With this version of gfortran, the installation of solcore fails.

I wonder therefore, if you haven't already tried it, whether you might try using one of the gfortran binary builds from [https://github.com/fxcoudert/gfortran-for-macOS/releases](https://github.com/fxcoudert/gfortran-for-macOS/releases)? (the install of gfortran in macports appends the version to the executable name so this doesn't conflict with the install in `/usr/local/bin`, however, if the gfortran executable in homebrew is named simply `gfortran` you'll probably want to set up your `PATH` to put `/usr/local/bin` first, (at least temporarily as a test, this may break other things if you've got multiple installs of tools in both `/usr/local` and via hombrew!)

phoebe-p, 2023-02-17:
@Chaz-Photonicity I am not sure if you gave up on using Solcore, but in case you are still interested: I am going to close this issue because with the new Solcore 5.9.0 (available on PyPI/through `pip install` since yesterday) the installation should now be much easier since you no longer need to compile the PDD on your own computer, regardless of the operating system. The documentation is updated, but you should just need to run `pip install solcore` (or `pip install solcore --upgrade` if you have an old version installed) to install with the PDD. Feel free to re-open the issue is you still have problems!

closed 2023-02-17 by phoebe-p