qpv-research-group/solcore5 #226: Using reference comparison with literal values

Abelarm, 2022-10-05:
While I was running the test on the PR #223 I noticed that in the case of test on `mac-os` some warning where generated where the reference comparison `is`/`is not` are used for literal values:

```
/home/runner/work/solcore5/solcore5/solcore/absorption_calculator/absorption_QW.py:148: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if line_shape is "Gauss":
/home/runner/work/solcore5/solcore5/solcore/absorption_calculator/absorption_QW.py:200: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if line_shape is "Gauss":
/home/runner/work/solcore5/solcore5/solcore/analytic_solar_cells/QE.py:137: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if layer.role is not 'emitter':
/home/runner/work/solcore5/solcore5/solcore/analytic_solar_cells/QE.py:152: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if junc[idx + 1].role is 'intrinsic':
/home/runner/work/solcore5/solcore5/solcore/analytic_solar_cells/QE.py:155: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if junc[idx + 2].role is 'base':
/home/runner/work/solcore5/solcore5/solcore/analytic_solar_cells/QE.py:165: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif junc[idx + 1].role is 'base':
/home/runner/work/solcore5/solcore5/solcore/poisson_drift_diffusion/QWunit.py:296: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if self.labels[index] is "well":
/home/runner/work/solcore5/solcore5/solcore/poisson_drift_diffusion/QWunit.py:305: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif self.labels[index] is "interlayer":
/home/runner/work/solcore5/solcore5/solcore/quantum_mechanics/high_level_kp_QW.py:79: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if "potentials" is graphtype:
/home/runner/work/solcore5/solcore5/solcore/quantum_mechanics/high_level_kp_QW.py:83: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if "potentialsLDOS" is graphtype:
/home/runner/work/solcore5/solcore5/solcore/quantum_mechanics/kp_QW.py:139: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if block is "U":
/home/runner/work/solcore5/solcore5/solcore/units_system/units_system.py:367: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  "%s: %s" % (dim, ", ".join([k for k in self.dimensions[dim].keys() if k is not None and k is not ""])))
```

**To Reproduce**
Go on `mac-os` check [here](https://github.com/qpv-research-group/solcore5/actions/runs/3184379247/jobs/5192830543) 

**Expected behavior**
Use `==` in case of literal values 

@dalonsoa if you think is worth to fix I can work on it ( really easy one)

dalonsoa, 2022-10-07:
Yes, please, go ahead with it. Solcore has a lot of legacy code that could do with some refactoring - this is one of the low hanging fruits, so pick it, for sure!

phoebe-p, 2022-10-14:
Fixed by #230

closed 2022-10-14 by phoebe-p