qpv-research-group/solcore5 #249: S4.new()

mosespv96, 2023-03-03:
when I try to use the RCWA to determine the EQE I encounter an issue with S4.New() as it says: 
AttributeError: module 'S4' has no attribute 'New'

phoebe-p, 2023-03-04:
Hi, could you give some more information - what operating system are you using? Did you encounter any issues during S4 installation?

are you trying to run a file from inside the same folder where your downloaded S4? If there is a local folder called “S4”, Python could be trying to import locally instead of from the installed package 📦

mosespv96, 2023-03-04:
When I try to run a file outside S4 I am unable to import S4. How do I install S4 as a package and call it in another file?
I tried to install S4 by:
make S4_pyext 
Is there something else we need to update?

phoebe-p, 2023-03-05:
Ok, you should be running it in a different folder (local importing will not work). Did you follow the installation instructions here? https://github.com/phoebe-p/S4

Did the installation seem to complete successfully? And what operating system are you using?

phoebe-p, 2023-03-05:
The instructions here https://rayflare.readthedocs.io/en/latest/Installation/installation.html may also be helpful

mosespv96, 2023-03-05:
Yes, I followed the same instructions. I am using ubuntu

phoebe-p, 2023-03-05:
What is the final bit of output in the terminal when you run `make S4_pyext`? Have you installed S4 and Solcore in the same Python environment you are using to run the code?

mosespv96, 2023-03-05:
I got something like S4 successfully installed with some version number. I am not sure if I am running the file in the same  python environment as Solcore and S4. How to check it?

mosespv96, 2023-03-05:
Thank you for the help. It was an issue with the environment and package installation and solved it with correct installation and python environment.

closed 2023-03-05 by mosespv96