qpv-research-group/solcore5 #250: Tutorial: 2J solar cell with QWs in the bottom cell

mosespv96, 2023-03-07:
Hi

Is it possible to share how you have got the plots for [Tutorial: 2J solar cell with QWs in the bottom cell](https://docs.solcore.solar/en/master/Examples/tutorial.html)?

phoebe-p, 2023-03-07:
I think this example makes all the plots. https://github.com/qpv-research-group/solcore5/blob/develop/examples/advanced_examples/MJ_solar_cell_tutorial.py

mosespv96, 2023-03-08:
So for the Isc we take the loglog and for the other three parameters Voc, FF and efficiency, we take the semilogx. Can you share the reason? How do we get power conversion efficiency of the solar cell? I am getting kind of similar values for Isc for each concentration and unexpected values for other parameters.

phoebe-p, 2023-03-09:
Taking the loglog vs. logx isn't related to Solcore, it's just a decision on how to plot the data. I guess it's because the Jsc will be a straight line on either a linear or log log scale since it's just proportional to concentration, while the Voc is proportional to log(concentration). 

As you can see in other examples, you can get the efficiency by solving for the light_iv and maximum power point and getting the my_solar_cell.iv["Eta"] attribute. In this case, the code is saving the power at the maximum power point instead (`my_solar_cell.iv["Pmpp"]`) so you can calculate the efficiency from that with the input power of the spectrum (which was depend on the concentration).

It is hard to answer your questions if you don't give a specific issue - I am not sure what "kind of similar values for Isc" or "unexpected values" means. I just ran the example and the results look reasonable; current is proportional to concentration, Voc is proportional to log(concentration), efficiency and fill factor increase with concentration up to the point that series resistance becomes an issue and then they start to decrease. Are you running the most recent version of this example/do you have the most recent version of Solcore installed? There was a problem with this example at one point (see #164) but it was fixed quite a while ago, I just ran it on my computer and the results look as expected.

phoebe-p, 2023-03-10:
This is very strange, thanks for posting the plots! I have never seen an issue like this before. The Jsc looks normal but there is clearly a problem with the Voc/FF and hence the efficiency. This does not happen when I run the example (I also checked on Ubuntu) so I'm not sure what's going on! Could you check which version of Solcore you have installed? Are you running the code exactly as in the example or did you modify anything? Could you also post the IV plot (the final one that gets plotted)?

mosespv96, 2023-03-10:
[image]
Sorry for the ambiguity in the plot figure. 

When I try solcore.config, first few lines are:
[Configuration]
version = 5.9.1
welcome_message = 1
verbose_loading = 1
....

Yes, I am trying the same example

phoebe-p, 2023-03-10:
@dalonsoa do you have any idea what could be happening here? I have no idea since I can't replicate this behaviour (on MacOS or on Ubuntu)...

mosespv96, 2023-03-10:
There was a small difference in temperature value and I am running it again, expecting to get the original plot

dalonsoa, 2023-03-10:
This is indeed very odd. I've never seen that before, either. 

@mosespv96 , could you confirm what **operative system** you are using, the **python version** as well as the version of **all python packages you have installed**? You can get a full list of the packages installed in your virtual environment, or system environment if you are not using a virtual one, with `pip list`. Even better, if you can do `pip freeze > solcore_dep.txt` and share here that file, it will help us to debug the issue. Equally, if you have modified the example, please share that file here, too.

If you are not using a virtual environment, I strongly recommend you to use it, as that isolate any potential conflicts with other python packages in your system. Check this [nice tutorial on the topic](https://realpython.com/python-virtual-environments-a-primer/).

phoebe-p, 2023-03-10:
> There was a small difference in temperature value and I am running it again, expecting to get the original plot

Let us know if that works, but even a different temperature should not produce these results! As Diego mentioned, I think it is worth trying to make a new virtual environment with just Solcore and its requirements and seeing if you still get the same thing.

mosespv96, 2023-03-10:
[image]
[image]
Now, I am getting somewhat similar to original plots. I am not sure if FF plot is little different than expected and how temperature is related to the efficiency and other parameters as it is affecting certain concentration values. Yes, I was running in a virtual environment.