qpv-research-group/solcore5 #255: InGaAs material data is incorrect

phoebe-p, 2023-04-04:
**Describe the bug**
The n/k data for InGaAs alloys (unless In fraction = 0) does not look correct, and the data files end at 840 nm even for the compositions where the bandgaps should be well beyond this wavelength.

**To Reproduce**

```
from solcore import material
import numpy as np
import matplotlib.pyplot as plt
from solcore.constants import q
import seaborn as sns

cols = sns.cubehelix_palette(8, start=.5, rot=-.75)

wl = np.linspace(300, 1000, 1000)*1e-9
In_c = np.linspace(0, 0.24, 5)

for i1, in_f in enumerate(In_c):
    GaAs = material("InGaAs")(In=in_f)
    plt.semilogy(wl*1e9, GaAs.k(wl), label=in_f, color=cols[i1])
    print(GaAs.get("band_gap")/q)

plt.legend(title="In fraction")
plt.xlabel("Wavelength (nm)")
plt.ylabel("k")
plt.show()
```

This gives the following plot:

[image]

Clearly, the weird oscillations near the bandgap are not correct. The raw data files look like this:

[image]

Which is less weird, but still problematic because they end at a very short wavelength. It seems that something is going wrong in the critical point interpolation (maybe because of the missing wavelengths?) which causes the strange near-bandgap behaviour.

phoebe-p, 2023-04-04:
@dalonsoa, do you have any idea where this optical constant data for InGaAs came from? There doesn't seem to be a note of it anywhere that I can find. There are four files, for 0%, 10%, 20% and 24% In.

dalonsoa, 2023-04-16:
Sorry - was on holidays. This data predates me. Maybe @iclned knows where it is coming from. It is also unclear if it is relaxed InGaAs or strained, grown on GaAs...

phoebe-p, 2023-04-18:
Hi Diego, no problem, I think I figured it out anyway! The data seems to come from the Ioffe database (not sure what their source is), but it is very low-quality and apparently for strained InGaAs which does not really make sense for a bulk junction! I have identified an alternative data source for relaxed InGaAs.