qpv-research-group/solcore5 #261: Cannot access data for GaAsN_zinc_blende

sbhasan, 2023-10-27:
Hi,

I am unable to extract the electrical properties of GaAsN_zinc_blende from the database. The error messages do not seem to be recognizable due to lack of familiarity with the code base. The error is easily reproduced by

import solcore as SC
SC.material('GaAsN_zinc_blende')(T=300, N=0.1).band_gap

This gives an error of being unable to find 'x'. I naively tried to modify parameter_system.py so as to replace 'x' in the string_expression variable with list(others.keys())[0] which in this instance is just 'N'. However, the results I get are unphysical. Either I still could not fix the actual problem or perhaps some coefficient entries are wrong.

Kind regards,
Shakeeb

phoebe-p, 2023-10-27:
Hi Shakeeb,
thanks for reporting this, there is definitely something wrong here. There is a typo in one of the material database files, specifically material_data/Vurgaftman/ternaries.txt. It currently reads:

```
[GaAsN_zinc_blende]
parent0=GaAs
parent1=GaN
x=N
Eg_Gamma=120.4-100.*x eV
spin_orbit_splitting=0 eV
```

These values are the bowing parameters for the various quantities (otherwise they are linearly interpolated between the two 'parent' materials, so GaAs and GaN in this case, and the line for Eg_Gamma should have `N` instead of `x`, which is the origin of the error you are getting. Changing `x` to `N` fixes the error:

```
Eg_Gamma=120.4-100.*N eV
```

 The equation for the bowing looks extreme, but checking the [Vurgaftman paper](https://pubs.aip.org/aip/jap/article/89/11/5815/488612/Band-parameters-for-III-V-compound-semiconductors) it does appear to be correct (Table XXX), but only or very small N fractions, since the bowing seems to be composition-dependent from reading the GaAsN section in that paper.

I wil fix this bug for the next release version of Solcore, thank you for pointing it out.

phoebe-p, 2023-11-20:
Typo fixed in 5.10.0 (though caveats about the validity of the bowing parameter remain).

closed 2023-11-20 by phoebe-p