qpv-research-group/solcore5 #262: Doc: Fortran compiler section needs update

yihuajack, 2023-11-03:
In http://docs.solcore.solar/en/latest/Installation/compilation.html#windows "Windows" subsection, the links [detailed instructions written by Michael Hirsch](https://www.scivision.co/f2py-running-fortran-code-in-python-on-windows/) and [Michael Hirsch](https://www.scivision.co/f2py-running-fortran-code-in-python-on-windows/) are no longer available.

phoebe-p, 2023-11-20:
@yihuajack this should be fixed now, version 5.10.0 has wheels up to Python 3.12 available through `pip install solcore`

lpdsamis, 2024-04-07:
This error appears when running the examples in version solcore 5.10 
warnings.warn('All voltages are positive, but junction has been identified as n-p,