qpv-research-group/solcore5 #263: Add support for ifort (Intel Fortran Compiler)

yihuajack, 2023-11-03:
According to https://community.intel.com/t5/Intel-Fortran-Compiler/ifort-command-line-warning-10161/td-p/822737 Intel Fortran Compiler does not support *.f95 file compilation, so it will failed to compile `solcore\poission_drift_diffusion\DDmodel-current.f95`. `ifor DDmodel-current.f95` gives
```text
Intel(R) Fortran Intel(R) 64 Compiler Classic for applications running on Intel(R) 64, Version 2021.9.0 Build 20230302_000000
Copyright (C) 1985-2023 Intel Corporation.  All rights reserved.

ifort: command line warning #10161: unrecognized source type 'DDmodel-current.f95'; object file assumed
ipo: warning #11010: file format not recognized for E:\Documents\Master\ECO\Simulation Programs\solcore5\solcore\poisson_drift_diffusion\DDmodel-current.f95
Microsoft (R) Incremental Linker Version 14.37.32825.0
Copyright (C) Microsoft Corporation.  All rights reserved.

-out:DDmodel-current.exe
-subsystem:console
DDmodel-current.f95
DDmodel-current.f95 : fatal error LNK1107: invalid or corrupt file: cannot read at 0x154A6.
```
`pip install solcore` gives
```text
Collecting solcore
  Using cached solcore-5.9.2.tar.gz (4.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [40 lines of output]
      + meson setup --prefix=D:\Program_Files\Python312 C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a\.mesonpy-jfudbxyr\build --native-file=C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a\.mesonpy-native-file.ini -Ddebug=false -Doptimization=2 --python.purelibdir D:\Program_Files\Python312\Lib\site-packages --python.platlibdir D:\Program_Files\Python312\Lib\site-packages
      The Meson build system
      Version: 1.2.3
      Source dir: C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a
      Build dir: C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a\.mesonpy-jfudbxyr\build
      Build type: native build
      Project name: solcore
      Project version: 5.9.2
      Host machine cpu family: x86_64
      Host machine cpu: x86_64
      Message: WINDOWS: true
      Program python3 found: YES (D:\Program_Files\Python312\python.exe)
      Message: py DIR D:\Program_Files\Python312\python.exe
      Message: install DIR D:/Program_Files/Python312/Lib/site-packages\
      ..\..\solcore\poisson_drift_diffusion\meson.build:4: WARNING: add_languages is missing native:, assuming languages are wanted for both host and build.
      C compiler for the host machine: icl (intel-cl 2021.9.0)
      C linker for the host machine: xilink.exe xilink 2021.9.0
      ..\..\solcore\poisson_drift_diffusion\meson.build:5: WARNING: add_languages is missing native:, assuming languages are wanted for both host and build.
      Cython compiler for the host machine: cython (cython 3.0.5)
      Fortran compiler for the host machine: ifort (intel-cl 2021.9.0)
      Fortran linker for the host machine: xilink.exe xilink 2021.9.0
      Compiler for C supports arguments -Wno-unused-but-set-variable: NO
      Compiler for C supports arguments -Wno-unused-function: NO
      Compiler for C supports arguments -Wno-unused-variable: NO
      Compiler for C supports arguments -Wno-surprising: NO
      Compiler for C supports arguments -Wno-conversion: NO
      Compiler for C supports arguments -Wno-misleading-indentation: NO
      Compiler for C supports arguments -Wno-incompatible-pointer-types: NO
      Compiler for Fortran supports arguments -Wno-unused-variable: NO
      Compiler for Fortran supports arguments -Wno-surprising: NO
      Compiler for Fortran supports arguments -Wno-conversion: NO
      Compiler for Fortran supports arguments -Wno-maybe-uninitialized: NO
      Compiler for Fortran supports arguments -Wno-unused-dummy-argument: NO
      Run-time dependency python found: YES 3.12
      Message: numpy DIR C:\Users\Yihua\AppData\Local\Temp\pip-build-env-piqp7323\overlay\Lib\site-packages\numpy\core\include
      Message: f2py DIR C:/Users/Yihua/AppData/Local/Temp/pip-build-env-piqp7323/overlay/Lib/site-packages/numpy/core/include/../../f2py/src

      ..\..\solcore\poisson_drift_diffusion\meson.build:74:7: ERROR: No host machine compiler for 'solcore/poisson_drift_diffusion/DDmodel-current.f95'

      A full log can be found at C:\Users\Yihua\AppData\Local\Temp\pip-install-5_wtxotf\solcore_3ceefe23c2a24f3e999aa976d8240a4a\.mesonpy-jfudbxyr\build\meson-logs\meson-log.txt
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

phoebe-p, 2023-11-03:
Thank you for reporting this issue. We changed the build system of Solcore so that it should no longer be necessary to compile it on the user's computer, but it was not yet updated to pre-build wheels for Python 3.12, which is why it was trying to compile the PDD solver on your computer. Note that I am also working on incorporating a Python-based PDD solver into Solcore which should be available in the next few weeks, and should not have any of these installation issues.

For now, there are two options:
1. If you are not actually planning to use the drift-diffusion solver, you can install Solcore without it by running:

```
pip install solcore --config-setting=setup-args="-Dwith_pdd=false"
```

2. If you do want to use it, I will add Python 3.12 to the build system and release a new version when I have confirmed it is working, and will update here when that is the case.

yihuajack, 2023-11-06:
@phoebe-p Thanks for your reply and your effort! I'd temporarily use Python 3.11 and wait for later releases:)

phoebe-p, 2023-11-20:
@yihuajack this should be fixed now, version 5.10.0 has wheels up to Python 3.12 available through `pip install solcore`

yihuajack, 2024-11-24:
> @yihuajack this should be fixed now, version 5.10.0 has wheels up to Python 3.12 available through `pip install solcore`

Hi @phoebe-p 
For solcore-5.10.0 `pip install solcore`, these errors still occur. Can you re-open this issue?

phoebe-p, 2024-11-25:
Sorry you are still having issues. Is the error exactly like what you posted above? I don't understand how there could be an issue, since there are now pre-built wheels for Python 3.12, no compilation should be happening when you install Solcore. Do you have an unusual combination of operating system and system architecture (like running Windows on a Mac computer)? I need some more information to figure out why `pip install solcore` is trying to compile rather than installing a wheel.

yihuajack, 2024-12-10:
> Sorry you are still having issues. Is the error exactly like what you posted above? I don't understand how there could be an issue, since there are now pre-built wheels for Python 3.12, no [...]

Thank you. Sorry I just find that Python 3.12 is OK; I was using Python 3.13 before

closed 2025-02-07 by phoebe-p