qpv-research-group/solcore5 #266: Strange behavior of PDD Fortran solver at high temperatures

sbhasan, 2023-11-26:
Hi,

I find the original Fortran based PDD solver to behave very strangely at high temperatures. The qualitative trend is there: Jsc increases and Voc decreases with increasing temperature. However, the quantitative difference with the experimental values is too large to bridge by tweaking any physical parameter. I have seen this for many different cells to think it is associated with a certain material.

For a long time I wondered if it was due to my inability to configure solcore right. However, the inclusion of the new Sesame solver gave me a new opportunity to quickly compare. To my surprise, changing the solver choice in existing models produced results that resemble very closely to experiments.

I have attached a notebook that works out an example. Here is a plot from it that shows the difference between results of the two solvers:

[image]

[pdd_vs_sesame_comparison.zip](https://github.com/qpv-research-group/solcore5/files/13468044/pdd_vs_sesame_comparison.zip)

Please let me know if I should share anything more to elaborate.

Kind regards,
Shakeeb

sbhasan, 2023-12-01:
Hi,

After some toiling, I managed to get it work as expected. I noticed two problems:

1) The iv_pdd function had an unintended (?) feature which prevented the value of T from propagating forward. I readjusted it so as to ensure that short-circuit/equilibrium calls get the right value of T.
2) I am uninitiated in Fortran so here I could be wrong. As far as I could see, the global variable T in the Fortran code was never updated. I modified the relevant functions in Fortran and the corresponding calls from Python to ensure the value of T is updated in Fortran.

And now, the comparison plot from the previous post looks as follows:

[image]

The updated files are attached. My edits can be easily found by searching for SBH 

[DriftDiffusionUtilities_updated.zip](https://github.com/qpv-research-group/solcore5/files/13520769/DriftDiffusionUtilities_updated.zip)

Thanks a lot for the wonderful work!

Kind regards,
Shakeeb

phoebe-p, 2023-12-07:
Thank you very much for reporting this and correcting the issue! I will check what the problem was/what you did and update it in the next version.

phoebe-p, 2024-03-15:
https://github.com/all-contributors please add @sbhasan for bug

phoebe-p, 2024-04-02:
@all-contributors please add @sbhasan for bug

phoebe-p, 2024-04-02:
Fixed by #273. Sorry this took a while to fix on my end, thanks again for identifying and fixing this issue!

PR #273 (2024-04-04): PDD and SPECTRAL2 fixes
Fixes #266 (thanks to Fortran code fixes by @sbhasan) and also fixes and issue in the spectral2.py file, where the "urban" AOD model, or c and d coefficients passed as lists by the user, could not be set due to a typo and incorrect type checking.
closed 2024-04-04 by phoebe-p