qpv-research-group/solcore5 #276: Tolerance in tests ?

Abelarm, 2024-04-15:
Hi @phoebe-p,

I keep getting annoying messages from Github about test failing in github action, for my fork of solcore.
All of them are because the tolerance is way too low.
I was asking is by design that solcore need this strict tolerance or we can relax it a bit to not make the test fail?

Examples:
[image]
[image]
[image]

phoebe-p, 2024-05-20:
Sorry, missed this somehow. We can definitely increase the tolerance on the tests, this is a silly reason to have failed tests.

Abelarm, 2024-06-08:
No worries, what's a good tolerance we like?

phoebe-p, 2024-06-27:
I've increased the tolerance for some tests I have seen failing due to this problem in #278. There are probably others that I missed. There were a couple where no tolerance was set in approx at all (so it's using the default of 1e-12 absolute/1e-6 relative), which I set to a relative tolerance of 1e-5. There were some other tests where a tolerance of 1e-4 was set which sometimes failed, changed those to 1e-3. let me know if it keeps happening with other tests!

closed 2025-02-07 by phoebe-p