qpv-research-group/solcore5 #283: TPV usage

aalfonsi, 2024-09-16:
**Please, confirm the following:**

- [x ] I have checked [Solcore’s documentation](http://docs.solcore.solar/en/master/) and could not find the answer.
- [x] I haven’t found any [other issue](https://github.com/qpv-research-group/solcore5/issues?q=) on a related topic.

**What’s the question?**

I was wondering if there is any experience of using solcore for TPV performance evaluation.  Overall my question is about the
possibility to model the recuperation effect of introducing a back reflector (e.g. gold) that allows for recycling the out-of-band spectrum (allowing re-absorption in the emitter).

**Additional information**
Add any other relevant information that could help us answering your question, E.g.:
- Code of a minimum working example showing what you’re trying to do.
- Screenshot showing the problem you are talking about.
- Context in which your question applies

phoebe-p, 2024-09-24:
I do not have experience using Solcore for TPV myself, but there is [this paper](https://pubs.aip.org/aip/adv/article/13/2/025263/2877658/Simulation-of-edge-effects-in-thermophotovoltaic) (I don't know the authors and wasn't involved in this work, so do not know have any details beyond what is reported in the paper). It's possible other people (not in academic research, so this would not be published) are using it for TPV calculations too but I am not sure.

In terms of whether I think Solcore is suitable for this, it can certainly simulate the optics of the reflector using the transfer-matrix method (assuming the structure is planar. If you have some kind of texture or grating, we also develop a more advanced optics package called [RayFlare](https://rayflare.readthedocs.io)), and the user can set the temperature of both the device and environment. Since Solcore was developed for PV rather than TPV, it's likely that the semiconductors you need to simulate are not in the existing database, and that some materials you want to use which are in the database do not have refractive index/absorption in the database up to sufficiently high wavelengths (e.g. the data for gold is from 0.2 to 2.5 um). However, you can add your own materials to the database as well. 

I hope this is helpful but please let me know if you have other questions!

alfoa, 2024-10-02:
@phoebe-p Thank you very much for your reply. I will give a try and let you know.

closed 2025-02-07 by phoebe-p